sebanc/brunch #809: lipnus lite boot failed when installed on hdd but works on USB

Checkmate95, 2021-01-12:
laptop : lenovo ideapad 100s 11-IBY (supported 64bit architecture but run on 32 bit uefi)
cpu: intel atom z3735f chromebook
image: sammus 87
brunch: Brunch r87 stable 20201227

there are 2 problems for this model
1.I successfully install brunch and chromeos to my sandisk extreme 32gb and everything work normally except screen brightness control not work and always set to 0% (almost black) but  if I log in and close lid to let laptop sleep then wake up screen brightness work again (finding for solution), I also tried install brunch with orco image (same model of laptop with same chipset and cpu) but these problem still appear. 

2.I tried to install chrome os to laptop's hdd which is emmc (I guess) from USB, I tried every method(from linux,windows and chrome os) install is normally success but when I boot there are error messages "lipnus lite boot failed" and another is "Default Boot Device Missing or Boot Failed. Insert Recovery Media and Hit anykey then Select 'Boot Manager' to choose a new Boot Device or to Boot Recovery Media"
my disk path is /dev/mmcblk1
I googled and found similar problem in same model in cloudready ([here](https://neverware.zendesk.com/hc/en-us/community/posts/360048471693-Lenovo-IdeaPas-100S-doesn-t-work)) I tried but doesn't work

sebanc, 2021-01-13:
1) I am not sure about the root cause of the brightness issue, could you try kernel 4.19 or 5.10 and see if the issue is still present ? (Replace "/kernel" in the grub config with "/kernel-4.19" or "/kernel-5.10")

2) it seems your device won't boot the default path /efi/boot/bootx32.efi by default on the hdd. You will have to manually create an efi entry that points to this path either directly in the bios or throught efibootmgr (from the brunch usb or through a Linux live flashdrive)

Checkmate95, 2021-01-13:
thank you sebanc you're awesome ! for brightness problem fixed by kernel-4.19 and efibootmgr for method 2 worked! 

note: 
1 .the only problem for now is built in 3.5mm headphone doesn't work ([temporary fixing by using external usb soundcard](https://github.com/sebanc/brunch/issues/803#issuecomment-757521584))
2. changing kernel to 4.19 after signed in may need to do powerwash (factory reset) bacause network doesn't work (found this problem from my brunch usb) this progress should be done after installation.
3. for creation boot in efibootmgr don't remove or replace default linpus lite, just create a new one and aim to bootia32.efi because replacing default linpus lite will make it ignore new one and revert to default after reboot.

Checkmate95, 2021-01-13:
Hi sebanc I just come here to update about brightness problem, I just figured out that it isn't bug from kernel about brightness control but my laptop model has turn off screen mode and just TURN SCREEN ON to let brightness work (OMG what I've tried to fix). BTW using kernel 5.4 and 5.10 will boot with screen-off mode instead screen-on mode like in kernel 4.19

sebanc, 2021-01-14:
Unfortunately, baytrail devices are known to have poor sound support in linux and additionally have headphone jack routing issues in chromeos.

Sound in chromeos works pretty much the same as in linux (except that it does not use pulseaudio), you can try some of the solutions you find in the linux forums (custom alsa commands, ucm files...).