sebanc/brunch #811: Pls add adapter driver for usb wifi dongle

truongtam-fudosan, 2021-01-14:
Realtek rtl8188eus & rtl8188eu & rtl8188etv WiFi drivers
https://github.com/aircrack-ng/rtl8188eus

Realtek RTL8811CU/RTL8821CU USB wifi adapter driver version 5.4.1 for Linux 4.4.x up to 5.x
https://github.com/brektrou/rtl8821CU

sebanc, 2021-01-14:
You can already use rtl8188eu driver by adding "options=rtl8188eu".

truongtam-fudosan, 2021-01-14:
> You can already use rtl8188eu driver by adding "options=rtl8188eu" on the kernel command line.

I tried but not working. It is rtl8188eus

sebanc, 2021-01-14:
Iirc, it's the exact driver you point that I implemented so I would need a dmesg to debug it (run "dmesg > ~/Downloads/dmesg.txt in crosh shell and attach dmesg.txt from your Downloads folder)

truongtam-fudosan, 2021-01-14:
[dmesg.txt](https://github.com/sebanc/brunch/files/5812845/dmesg.txt)

truongtam-fudosan, 2021-01-14:
> Iirc, it's the exact driver you point that I implemented so I would need a dmesg to debug it (run "dmesg > ~/Downloads/dmesg.txt in crosh shell and attach dmesg.txt from your Downloads folder)

[  146.072652] r8188eu: module is from the staging directory, the quality is unknown, you have been warned.
[  146.084793] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)

truongtam-fudosan, 2021-01-14:
Test in Linux Mint working.
https://www.tp-link.com/en/support/download/tl-wn725n/
http://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=13&id=128

sebanc, 2021-01-15:
I am a bit confused as there are 2 wifi adapters found in your dmesg (r8188eu and r8723bs). Are both plugged in your computer ?

Could you run:
`echo "blacklist r8188eu" | sudo tee /etc/modprobe.d/rtl.conf`
`echo "blacklist r8723bs" | sudo tee -a /etc/modprobe.d/rtl.conf`

and post another dmesg ?

truongtam-fudosan, 2021-01-15:
> I am a bit confused as there are 2 wifi adapters found in your dmesg (r8188eu and r8723bs). Are both plugged in your computer ?
> 
> Could you run:
> `echo "blacklist r8188eu" | sudo tee /etc/modprobe.d/rtl.conf`
> `echo "blacklist r8723bs" | sudo tee -a /etc/modprobe.d/rtl.conf`
> 
> and post another dmesg ?

I tried install but can not load. `Unknown symbol __stack_chk_guard (err -2)` build in kernel 4.19 and 5.4 both.
r8723bs is wireless machine. I check `sudo lshw -C network` -> not show USB device.

sebanc, 2021-01-17:
I tried to fix "options=rtl8188eu" in the latest brunch-unstable build and added "options=rtl8821cu", could you try them and post a dmesg if they do not work ?

truongtam-fudosan, 2021-02-01:
@sebanc Thank you very much! It working.

closed 2021-02-01 by truongtam-fudosan