sebanc/brunch #813: Chrome OS ANDROID APPS NOT WORKING!!!

Pavel189, 2021-01-15:
Hello, I'm new to the brunch community and have recently encountered an error with Android apps and Google Play Store. Whenever I try to launch it, the icon becomes black and white and there is a spinning circle in it. When I freshly installed it, the Play Store did not work, but on the other day after a lot of reboots it worked. But now it doesn't work again, and I have tried the following things but it still didn't work:

Disabled Google Sync
Tried deleting the app data (The setitngs app for android simply did not open, it's like the whole android is not working)
Removing and re-enabling the Play Store (all it did is just delete all of my apps)
Re-signing into my Google account

I'm running release 86 on brunch, and using samus recovery image, here are my laptop specs:

HP 15 Notebook PC
Intel Pentium N3540
4GB RAM
Intel HD Graphics

Please help me, because I really like Chrome OS. ans want to get the Android apps working...

Sincerely, 
Pavel
[image]

khanhtranngoccva, 2021-01-16:
BRUH just wait for a minute on the login screen. Or just log out and log back in again if you already have. This only happens on an HDD/USB, not on SSD or NVMes that are much faster and ChromeOS is supposed to run on. 
If you want to know when to log in anyway (like I do), just have this simple script, copy it to the system partition, and just run it in the console view using Ctrl + Alt + F2 and user "chronos": (return to main screen with Ctrl + Alt + F1). Change the extension to .sh and execute it with BASH.
[script.txt](https://github.com/sebanc/brunch/files/5823756/script.txt)

truongtam-fudosan, 2021-01-16:
[image]
Pls turn on Google Play Store

thehalfnotes, 2021-01-22:
> BRUH just wait for a minute on the login screen. Or just log out and log back in again if you already have. This only happens on an HDD/USB, not on SSD or NVMes that are much faster and ChromeOS is [...]

How to Execute this script ? * please elaborate *

b4iterdev, 2021-02-10:
as he said, copy it to the system partition then open console run 
```
mv script.txt script.sh
```
now reboot, when your device reboot at login screen, press Ctrl + Alt + F2. signin with username: chronos ( it already show on motd) then run:
```
bash script.sh
```
when it said:"You can now login to ChromeOS and run Android apps without problems.", press ctrl+alt+f1 then login. Enjoy!

sebanc, 2021-02-10:
Could someone try to run "sudo tune2fs -o journal_data_writeback /dev/sdX3" and see if it helps ? (replace "SdX" by your hdd)

b4iterdev, 2021-02-10:
> Could someone try to run "sudo tune2fs -o journal_data_writeback /dev/sdX3", reboot and and see if it helps ? (replace "sdX" by your hdd)

run it when logged in or tty ?

b4iterdev, 2021-02-10:
and mine is /dev/sda or /dev/sda9 (sda9 is chromeos partition), what should I choose ?

b4iterdev, 2021-02-10:
[image]
Not working ?

sebanc, 2021-02-10:
/dev/sda3

sebanc, 2021-02-10:
If you are in dual boot: "sudo tune2fs -o journal_data_writeback /dev/loop15p3"

b4iterdev, 2021-02-10:
[image]
Seem good, I will try to reboot and login immediately.

b4iterdev, 2021-02-10:
[image]
@sebanc Not working

b4iterdev, 2021-02-10:
[image]

logout then login again, it load!

b4iterdev, 2021-02-10:
dmesg log on this session: https://pastebin.com/cLbS1Afu

sebanc, 2021-02-10:
Ok, thanks for trying it.

While you are trying things, could you try the below kernel command line parameters (try one at a time):
- "elevator=none"
- "elevator=mq-deadline"
- "elevator=kyber"
- "elevator=bfq"

b4iterdev, 2021-02-10:
Ok, you can take a look at issue #864 which there are some infomation about what am I running

b4iterdev, 2021-02-10:
@sebanc none of them works, here is dmesg output of elevator=bfq, 3 others are the same, android container doesn't event mount: 
[image]
btw could you explain what does elevator means ?

b4iterdev, 2021-02-10:
@sebanc 
This line is output-ed when I logout: 
``` [  480.928725] capability: warning: `main' uses 32-bit capabilities (legacy support in use) ```
And these 3 lines are output-ed when I log back in: 
```
[  501.048978] ESDFS-fs (esdfs): mounted on top of /opt/google/containers/android/rootfs/android-data/data/media type ext4
[  501.049180] ESDFS-fs (esdfs): mounted on top of /opt/google/containers/android/rootfs/android-data/data/media type ext4
[  501.049380] ESDFS-fs (esdfs): mounted on top of /opt/google/containers/android/rootfs/android-data/data/media type ext4
```

sebanc, 2021-02-10:
Thanks for the logs, however I have already been through those before and they are not very informative.

Unfortunately it is difficult to debug this issue as I cannot reproduce it and I did not manage to delay the android init.

IMO the best chance to fix this is to find a way to modify the hdd scheduler or i/o settings in order to prevent the issue from occurring.

Could you try to comment the lines starting with "vm." in /etc/sysctl.d/00-sysctl.conf and reboot ?

b4iterdev, 2021-02-11:
Line commented:
vm.overcommit_memory = 1
vm.dirty_background_ratio = 5
vm.dirty_expire_centisecs = 60000
vm.dirty_ratio = 60
vm.dirty_writeback_centisecs = 60000
vm.laptop_mode = 0
vm.page-cluster = 0

b4iterdev, 2021-02-11:
Update: Not working, should I un-comment these lines?
[image]

b4iterdev, 2021-02-11:
I get this script from @khanhtranngoccva, with some changes that will exclusively let you know when you should login if this line are output-ed:
```
capability: warning: `main' uses 32-bit capabilities (legacy support in use)
```
## Guide:
Download this script (credit goes to @khanhtranngoccva ) : 
[apps.txt](https://github.com/sebanc/brunch/files/5962584/apps.txt)
Then press ctrl + alt + t and type ``` shell ``` , type ``` cd /home/user/[TAB]/Downloads```
Type ``` ls ``` and make sure there are apps.txt in.
Now type this command to move apps.txt to root directory and rename it to .sh
```
sudo mv apps.txt /apps.sh
```
When you reboot the system, press ctrl + alt + f2 to go to tty2:
When it asks your user login, type ``` chronos ``` with blank password and type:
```
bash /apps.sh
``` 
After this line are output-ed: "You can now login to ChromeOS and run Android apps without problems" just type "exit" (to logout of tty2) and press ctrl + alt + f1 and login.
At this time android apps should work, enjoy!

sebanc, 2021-02-11:
If someone is still interested to try to debug this, could he try to replace "start on starting failsafe" by "start on stopped udev-trigger-early" in /etc/init/udev-trigger.conf ?

sebanc, 2021-02-11:
or replace "start on continue-arc-boot" by "start on stopped udev-trigger" in /etc/init/arc-boot-continue.conf

b4iterdev, 2021-02-12:
I always interested to try this 👯

b4iterdev, 2021-02-12:
Should I do both or just one-by-one ?

b4iterdev, 2021-02-12:
> If someone is still interested to try to debug this, could he try to replace "start on starting failsafe" by "start on started udev-trigger-early" in /etc/init/udev-trigger.conf ?

Not working, system didn't finish post-boot before user login

sebanc, 2021-02-12:
Ok, when it's like that you can press "e" quickly on grub screen and add "options=rebuild" to recover the rootfs

WitaloA, 2021-07-29:
friend help me!!!!!!!!

srinivsn396, 2022-04-23:
HDD system can't open apk and Linux files or is there any specific verson for HDD laptop .. please help me..keep on spinning all of my apk and Linux files..

bybenjamin, 2022-11-26:
I'm getting the same error. Can anyone help? 
dmesg output

`[    0.000000] microcode: microcode updated early to revision 0x24000023, date = 2022-02-19
[    0.000000] Linux version 5.15.73-brunch-sebanc (chronos@localhost) (gcc (Ubuntu 12.1.0-2ubuntu1~22.04) 12.1.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP PREEMPT Sat Oct 29 12:32:56 UTC 2022
[    0.000000] Command line: BOOT_IMAGE=(hd0,7)/kernel boot=local noresume noswap loglevel=7 cros_secure cros_debug options= chromeos_bootsplash= console= vt.global_cursor_default=0 brunch_bootsplash=default quiet
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 1520
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000006a9f8fff] usable
[    0.000000] BIOS-e820: [mem 0x000000006a9f9000-0x000000006b2f8fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000006b2f9000-0x0000000073f0efff] usable
[    0.000000] BIOS-e820: [mem 0x0000000073f0f000-0x000000007534efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007534f000-0x0000000075b7efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000075b7f000-0x0000000075bfefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000075bff000-0x0000000075bfffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000075c00000-0x000000007f7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe010000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000001807fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by INSYDE Corp.
[    0.000000] efi: ACPI=0x75bfe000 ACPI 2.0=0x75bfe014 TPMFinalLog=0x75b46000 SMBIOS=0x744c2000 MEMATTR=0x6ff82018 ESRT=0x70223418 RNG=0x744c0f98 TPMEventLog=0x6b30a018 
[    0.000000] efi: seeding entropy pool
[    0.000000] random: crng init done
[    0.000000] SMBIOS 3.3 present.
[    0.000000] DMI: Acer Swift SF114-34/Labatt_JL, BIOS V1.11 08/02/2021
[    0.000000] tsc: Initial usec timer 10035487
[    0.000000] tsc: Detected 1100.000 MHz processor
[    0.000000] tsc: Detected 1113.600 MHz TSC
[    0.000009] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000014] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000022] last_pfn = 0x180800 max_arch_pfn = 0x400000000
[    0.000085] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000395] last_pfn = 0x75c00 max_arch_pfn = 0x400000000
[    0.004219] esrt: Reserving ESRT space from 0x0000000070223418 to 0x0000000070223450.
[    0.004226] e820: update [mem 0x70223000-0x70223fff] usable ==> reserved
[    0.004559] Secure boot disabled
[    0.004560] RAMDISK: [mem 0x3980f000-0x3a810fff]
[    0.004566] ACPI: Early table checksum verification disabled
[    0.004570] ACPI: RSDP 0x0000000075BFE014 000024 (v02 ACRSYS)
[    0.004575] ACPI: XSDT 0x0000000075BD8188 00010C (v01 ACRSYS ACRPRDCT 00000002      01000013)
[    0.004583] ACPI: FACP 0x0000000075BFD000 00010C (v06 ACRSYS ACRPRDCT 00000002 1025 00040000)
[    0.004589] ACPI: DSDT 0x0000000075B98000 03B3D7 (v02 ACRSYS ACRPRDCT 00000002 1025 00040000)
[    0.004594] ACPI: FACS 0x0000000075B25000 000040
[    0.004598] ACPI: UEFI 0x0000000075B7E000 000236 (v01 ACRSYS ACRPRDCT 00000001 1025 00040000)
[    0.004603] ACPI: SSDT 0x0000000075BF8000 0049E2 (v02 ACRSYS ACRPRDCT 00001000 1025 00040000)
[    0.004607] ACPI: SSDT 0x0000000075BF6000 001CA4 (v02 ACRSYS ACRPRDCT 00003000 1025 00040000)
[    0.004611] ACPI: SSDT 0x0000000075BF2000 003291 (v02 ACRSYS ACRPRDCT 00003000 1025 00040000)
[    0.004616] ACPI: SSDT 0x0000000075BEF000 0025ED (v02 ACRSYS ACRPRDCT 00003000 1025 00040000)
[    0.004620] ACPI: SSDT 0x0000000075BEE000 00077B (v02 ACRSYS ACRPRDCT 00001000 1025 00040000)
[    0.004624] ACPI: TPM2 0x0000000075BED000 00004C (v04 ACRSYS ACRPRDCT 00000002 1025 00040000)
[    0.004629] ACPI: MSDM 0x0000000075BEC000 000055 (v03 ACRSYS ACRPRDCT 00000001 1025 00040000)
[    0.004633] ACPI: SSDT 0x0000000075BE9000 002FCB (v02 ACRSYS ACRPRDCT 00000000 1025 00040000)
[    0.004638] ACPI: NHLT 0x0000000075BE8000 00002D (v00 ACRSYS ACRPRDCT 00000002 1025 00040000)
[    0.004642] ACPI: SSDT 0x0000000075BE7000 0007DA (v02 ACRSYS ACRPRDCT 00001000 1025 00040000)
[    0.004646] ACPI: LPIT 0x0000000075BE6000 000094 (v01 ACRSYS ACRPRDCT 00000002 1025 00040000)
[    0.004651] ACPI: WSMT 0x0000000075BE5000 000028 (v01 ACRSYS ACRPRDCT 00000002 1025 00040000)
[    0.004655] ACPI: SSDT 0x0000000075BE4000 000B70 (v02 ACRSYS ACRPRDCT 00001000 1025 00040000)
[    0.004659] ACPI: SSDT 0x0000000075BE3000 000122 (v02 ACRSYS ACRPRDCT 00000000 1025 00040000)
[    0.004664] ACPI: DBGP 0x0000000075BE2000 000034 (v01 ACRSYS ACRPRDCT 00000002 1025 00040000)
[    0.004668] ACPI: DBG2 0x0000000075BE1000 000054 (v00 ACRSYS ACRPRDCT 00000002 1025 00040000)
[... cut ...]
[  105.542323] ESDFS-fs (esdfs): mounted on top of /opt/google/containers/android/rootfs/android-data/data/media type ext4
[  105.542443] ESDFS-fs (esdfs): mounted on top of /opt/google/containers/android/rootfs/android-data/data/media type ext4
[  833.271763] ESDFS-fs (esdfs): mounted on top of /opt/google/containers/android/rootfs/android-data/data/media type ext4
[  833.271928] ESDFS-fs (esdfs): mounted on top of /opt/google/containers/android/rootfs/android-data/data/media type ext4
[  833.272079] ESDFS-fs (esdfs): mounted on top of /opt/google/containers/android/rootfs/android-data/data/media type ext4`

GhostxPiercer, 2024-05-06:
hey im new to brunch chrome  and im having issues with the android apps.All of them are having a network issue idk why but evvery android doesnt work it say network error
[image]
[image]