sebanc/brunch #818: Anyway to blacklist a Nvidia GPU in brunch and use INTEL on board?

zenn87, 2021-01-20:
Praying to sebanc the god of chrome os here...

First of Sebanc thanks to you im one step closer to my OS dreams got my lenovo working thanks to you...

I have realized i cant completely ditch other ose's all together due to some gaming I wish to do. So im looking to dualboot lINUX And chrome os via brunch. Using two separate SSDS easy thanks to your work Sebanc.

But i would like to leave my gpu in my desktop, was hoping there was some magic to blacklist the nvidia card all together so i can use it in linux for gaming and use the onboard on mobo for chrome. Most of my workflow is now in chrome os, but cant drop a few games.

Ideally I would like to be able to run the hdmi from my gt 1030 to the hdmi input on my monitor and the the display port from the mobo to the other.

IF you have any suggestions or God powers to suggest sebanc I would be eternally grateful. This is the last step to me completely ditching windows full time which makes me excessively giddy.

sebanc, 2021-01-20:
You should not have to remove the nvidia gpu from your desktop for chromeos to work. What happens if you boot chromeos with the nvidia gpu in your desktop ?

The main issue I can see is that the HDMI port of your nvidia card will most likely not work and I have no solution for that, I would suggest using an usb-c hub for external displays.

zenn87, 2021-01-20:
Ithe video card is a gt 1030 , I haven't even tried to boot yet figured I consult you first. The desktop doesn't have usb connectivity at all.

closed 2021-01-20 by zenn87