sebanc/brunch #819: Boot issue

mhbear14, 2021-01-20:
I downloaded the latest brunch release
Brunch r87 stable 20201227 and updated using sudo chromeos-update command at first I thought the command was not running so I ran the command twice but I was wrong  the first command started running and updated it and then after that the second command took action I wanted to stop the second command from running by closing the window but I did not as I did not want anything to go wrong so after everything was completed I rebooted and now chrome os doesn't boot[]
I have added images in the drive
(https://drive.google.com/folderview?id=1TJ0pygjNOd2GKUbsbIR49u_rZdTN3-89)

ameyjoshi3068, 2021-01-20:
There is a way to fix any corrupt files in Chrome OS.  For that, you will need Linux, either you will need a live USB or Linux installed as dual boot. (I am not a Linux genius but I have used this procedure to fix the same problem).

1. Extract rootc.img from the downloaded brunch archive and mount it(normally can be done by right click and selecting mount). Copy all the contents in it.
2. First locate the chromeOS image file with file manager, then create a loop device by running `sudo kpartx -av <name of chromeos.img file>`
3. create a new directory by `sudo mkdir -p /mnt/tmpmount` 
4. mount partition no 7 of chromeos.img `sudo mount /dev/mapper/loopxp7 /mnt/tmpmount` where x is your loop device number.
5. Now open the tmpmount folder from file manager and don't forget to open it as root.
6. First delete all the contents in it and then paste all the copied files inside the folder.
7. Unmount the partition by running `sudo umount /mnt/tmpmount`
Now you can restart. Your chromeos rootfs will be rebuilt and your problem will be fixed.

omkar-panigrahi, 2021-01-21:
where do

> I downloaded the latest brunch release
> Brunch r87 stable 20201227 and updated using sudo chromeos-update command at first I thought the command was not running so I ran the command twice but I was [...]

Where Do You Enter sudo chromeos-update where does it should be entered i am comfused

ameyjoshi3068, 2021-01-23:
@omkar-panigrahi just press `Ctrl+Alt+T` then terminal will open
Now enter `shell` Now you can enter the code