sebanc/brunch #822: Broadcom BCM43142 not working properly(stuck on searching for wifi)

nawker, 2021-01-21:
I am dual booting brunch with windows 10 using grub. at first WiFi wouldn't turn on at all. but after adding "options=broadcom_wl" to the grub config. WiFi turns on now but keeps on searching and not finding a single WiFi network. I've seen other people reporting that the broadcom BCM43142 card did work for them. but i cant see where the issue is since the card works well on windows 10. i hope it can be solved because right now I am connecting to the internet via USB tethering from my android phone which isn't practical at all

my laptop is a Dell inspiron 15 3542
and my network card is: Broadcom BCM43142

is case it helps this is my slpci list
`chronos@localhost / $ sudo lspci
00:00.0 Host bridge: Intel Corporation Haswell-ULT DRAM Controller (rev 0b)
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 0b)
00:14.0 USB controller: Intel Corporation 8 Series USB xHCI HC (rev 04)
00:16.0 Communication controller: Intel Corporation 8 Series HECI #0 (rev 04)
00:1b.0 Audio device: Intel Corporation 8 Series HD Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 1 (rev e4)
00:1c.2 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 3 (rev e4)
00:1c.3 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 4 (rev e4)
00:1d.0 USB controller: Intel Corporation 8 Series USB EHCI #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation 8 Series LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 8 Series SATA Controller 1 [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 8 Series SMBus Controller (rev 04)
06:00.0 Network controller: Broadcom Corporation BCM43142 802.11b/g/n (rev 01)
07:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8101E/RTL8102E PCI Express Fast Ethernet controller (rev 07)`

sebanc, 2021-01-22:
To be honest, I don't really understand what's wrong with the broadcom_wl module, I used different sources to build the modules and it always work for some but not for others.

What I would need is for someone to try different kernel sources and tell me which one works best:
- Install the "brunch-toolchain"
- Download the source of the broadcom_wl module of your choice
- Run "make" in the source tree
- Copy the resulting wl.ko to /lib/modules/<kernel_name>/broadcom_wl.ko

If you find a source that works, I will try to switch to it.

jhs1004, 2021-01-25:
I would love to test but since I am not very techy, I would need a detailed guide, preferably with images... That said, in response to OP, try using kernel 5.10. My wifi stops working after going into sleep mode but it works after reboot. USB wifi dongles also work fyi.

jhs1004, 2021-01-25:
@sebanc, wrong thread to mention this but I think disabling security feature through hyperthreading seems to have solved overheating issue I've been having on my Macbook Pro 15 2013 late ver, although using the experiemental kernel 5.10 seems to help alot too. My fans are no longer on for basic tasks.