sebanc/brunch #823: rtl8812au doesn't work with 5GHz network

meciu, 2021-01-21:


sebanc, 2021-01-22:
Could you try with kernel 5.10 ? (replace "/kernel" in your grub config with "/kernel-5.10")

meciu, 2021-01-22:
Thank You for answer.
So I've tested if 5GHz shows up with 5.10 kernel but still no luck.

sebanc, 2021-01-22:
Could you try the release in brunch-unstable repo ?

meciu, 2021-01-25:
Tested **Brunch r87 unstable 20210124** and I'm unable to turn on WiFi.

dmesg attached
[dmesg.txt](https://github.com/sebanc/brunch/files/5867892/dmesg.txt)

sebanc, 2021-01-26:
I changed the driver, could you try the new brunch-unstable ?

madhits, 2021-01-27:
I'm seeing this issue on 3x BCM cards, rtl cards and ar5b97 and ar5b125 cards. Only intel has 5ghz. I'm assuming all these cards would use a different driver correct? but why would all the drivers show the same issue?

meciu, 2021-01-29:
With latest Unstable Brunch the WiFi cannot be turned on.
Will provide dmesg dump later

sebanc, 2021-01-31:
What's the ouput of `insmod /lib/modules/$(cat /proc/version |  cut -d' ' -f3)/rtl8812au.ko` command in crosh shell ?

sebanc, 2021-01-31:
Could you try the new stable brunch ?

meciu, 2021-02-02:
Okay,
With r88 wifi works, but still no 5GHz networks visible.
Tried command with insmod and result is
"Could not insert module ... rtl8812au.ko:File exists"

sebanc, 2021-02-07:
Could you try the build in the "brunch-unstable" repository ? Not sure it will change something but I have updated the wireless regulatory db

meciu, 2021-02-10:
Thanks for the update, but still the issue persists :(

closed 2026-03-19 by github-actions[bot]