sebanc/brunch #824: Phone hub is not shown up

QuanTranPPO, 2021-01-22:
Hello,
I am trying the new [Phone Hub](https://www.androidpolice.com/2020/12/14/this-is-phone-hub-for-chrome-os-googles-effort-to-tightly-link-android-and-chromebooks/?amp) feature in chrome://flags. However, it's not showing up in the shelf as intended. Is there any way to fix this?

ameyjoshi3068, 2021-01-23:
@QuanTranPPO Can you try going to dev or canary release?
P.S. I am also eagerly waiting for the feature to start working.

QuanTranPPO, 2021-01-23:
@ameyjoshi3068 I tried with Canary channel, but the phone hub icon didn't appear on the shelf neither. I can search "Phone Hub" in setting page, and it shows some options for Phone hub. But when I clicked on those, it would bring me to the " Connected device" section without any phone hub option.

sebanc, 2021-02-03:
I think the answer has just been posted on reddit:
https://www.reddit.com/r/chromeos/comments/lbqef4/how_can_i_enable_phone_hub_beta_89/

QuanTranPPO, 2021-02-05:
Yes. I guess we have to wait for Google rolling out to all users

ameyjoshi3068, 2021-02-07:
I finally got play services beta. Now waiting for server-side support.