sebanc/brunch #825: lenovo Thinkpad 10 (1st gen) auto-rotation, wacom pen, bluetooth audio not working at brunch_r87, chromeos_13505.73.0_rammus

chaejihwan, 2021-01-24:
Hi, Thank you for making it  

I'm using old lenovo thinkpad 10 tablet model 

it have atom z3775 processor and have 4gb ram, emmc 128gb drive, broadcom WIFI,BT, 

everything work fine. and performance moderately 

but my model not working auto-ratation,

and stylus pen working except for Pressure detect

Could you fix it? or I try another way?

I am bad at English. Thank you for reading through.

sebanc, 2021-01-24:
Could you try to switch to tablet mode manually with CTRL+ALT+SHIFT+T and then see if rotation works ?

chaejihwan, 2021-01-24:
> Could you try to switch to tablet mode manually with CTRL+ALT+SHIFT+T and then see if rotation works ?

Thank you sebanc it work fine 

i want always tablet mode 
 
option sysfs_tablet_mode / force_tablet_mode

 Which one should I choose between the two?

sebanc, 2021-01-24:
I think "force_tablet_mode" option is the one you want

chaejihwan, 2021-01-25:
> I think "force_tablet_mode" option is the one you want

thanks 👍 i solved 

good day!!!

lucasml1603, 2021-02-01:
HI, @chaejihwan i have the same Atom procesor, but don´t have audio and mic, Have you any configuration special or option aplied in Grub loader? I have "baytrail_chromebook", but nothing happend. Thanks.