sebanc/brunch #828: "Loading Brunch Framework" bootloop

MicroMihai, 2021-01-25:
Installed the system from Linux Mint, rebooted and "Loading Brunch Framework" has started. It rebooted by itself and I thought it was normal. When after 3/4 reboots I notice that it reboots the same way without changing or completing anything.
I've tried looking for something to fix this, but none apparently have a solution.

sebanc, 2021-01-25:
Sorry but as mentionned in the readme this amd cpu/gpu is not compatible

kushmittal2009, 2021-01-27:
That happened to one of my device to but using another recovery did the trick.

thehalfnotes, 2021-02-03:
reference to the #843

thehalfnotes, 2021-02-03:
I also use AMD processor

kissunyeason, 2021-03-29:
I also encountered the same problem, but I can start normally using cloudready.

kissunyeason, 2021-03-29:
> That happened to one of my device to but using another recovery did the trick.

Can you tell me which method you use? Or which video to refer to?

AdityaBhati1, 2021-07-18:
would anyone suggested me another recovery exept zork recovery ?