sebanc/brunch #831: 5ghz wifi not found in the Samus recovery with brunch

madhits, 2021-01-27:
I have tried 2x Arthos cards and 1 ratink card and I can not get 5ghz networks to show up, despite all these cards supporting 5ghz wireless N. I tried an Intel card and 5ghz does show up but the card has no Bluetooth. So I want to keep the Dell 1705 card which includes Bluetooth and so does the ralink. I saw another user has this issue but related it to dual booting and fixed by updating grub. Im only single booting brunch. Thoughts?

sebanc, 2021-01-31:
I don't think there is a general issue in chromeos/brunch with 5ghz, in general wifi support mostly depends on the support in the linux kernel for you wifi cards and also on the firmware used.

BCM cards are notably known to have no 5ghz support or may require specific firmware files. For the other cards, it is hard to say like that without knowing what kernel module is used.

madhits, 2021-02-01:
So what can I do?

sebanc, 2021-02-01:
I would suggest to:
- Identify the kernel modules used by your adapters
- Check if 5ghz is supported in the linux kernel with those modules.
- Verify if there are specific drivers included in brunch for those adapters (to be enabled with options)

If not, provide the reference of the adapter that's supposed to have the best support in linux and I will try to have a look at it.

madhits, 2021-02-01:
Ok, i'm still a bit of a linux noob. lspci should give me the kernal modules loaded/being used right? 

I can research from there the support and what drivers are being used.

Does that sound right?

THANKS

sebanc, 2021-02-07:
Could you try the build in the "brunch-unstable" repository ? Not sure it will change something but I have updated the wireless regulatory db

madhits, 2021-02-19:
OK going to try that. WIll get back to you when I try it, might be a week or so.