sebanc/brunch #834: Accidentally updated to Chrome OS 88 On Brunch 87 Stable

DeStRoYD, 2021-01-27:
Hi, I bought a brunch enabled "chromebook" around 4 months ago and it came with Chrome OS 85, the person I bought it from told me that it would auto-update when a new version was released, I waited until today, when I ran out of patience and I updated it using the manual mode from this person's video: https://www.youtube.com/watch?v=db4Xo_xPTR4
I used the latest brunch stable r87 20201227, thinking it would update to Chrome OS 87, but it actually updated it to Chrome OS 88
I wanted to know are there going to be any problems if I use the brunch r87 kernel with Chrome OS 88?
Here are my laptop specs:

Prestigio Smartbook 133S
Intel Celeron N3350
Intel HD Graphics
3GB RAM
Using "rammus" recovery image

Regards, 
Evgeni

madhits, 2021-01-27:
88 should work and I would not worry about it if you dont see any problems. 87 was a huge jump for chrome os and 88 is a minor upgrade. I would upgrade brunch when you see an update that supports 88 but until then enjoy.

clacalderonc, 2021-01-29:
I have no issues at this time updating to last release of chromeos, but. Why did you open an issue for this?. Better to keep the supported release, and if you want to try the last one, just keep in mind that there could be problems.

DeStRoYD, 2021-01-30:
Thanks for the responses, it seems fine until now and I will update to Brunch 88 when it's available, thanks again, I'm closing the issue!

closed 2021-01-30 by DeStRoYD