sebanc/brunch #835: Restart loop

mesmaili151, 2021-01-29:
Hi, i tried to install Chrome os dual boot.
after installing with linux and config boot with win2grup when i want to login in my chromos, i get this message and my laptop will restart automatically.

Loading brunch Framework...
ChromeOS rootfs patches are being applied.

my cpu is amd ryzen 7 and i tried to instarr ChromeOS grunt

ameyjoshi3068, 2021-01-29:
very few AMD CPUs are compatible with Chrome OS since there not exist many AMD Chromebooks.
Also, only intel graphics are compatible with Chrome OS. Just make sure you have one.
You can also try Zork recovery image and 4.19 or 5.10 kernels.

mesmaili151, 2021-01-29:
> very few AMD CPUs are compatible with Chrome OS since there not exist many AMD Chromebooks.
> Also, only intel graphics are compatible with Chrome OS. Just make sure you have one.
> You can also try Zork recovery image and 4.19 or 5.10 kernels.

Thanks @ameyjoshi3068 . also i tried to install Zork, but at this page ( ChromeOS rootfs... ) the page stops and does not change.

mesmaili151, 2021-01-29:
also my gpu is NVIDIA, i think i dont have any chance

ameyjoshi3068, 2021-01-30:
Sadly, Chrome OS won't work without intel graphics.

ameyjoshi3068, 2021-01-30:
Sadly, Chrome OS won't work with Nvidia GPU 
But if you also have AMD GPU (dGPU)  then there are chances that blacklisting Nvidia GPU will let you boot.
Try zork because it is made for ryzen cpus

mesmaili151, 2021-02-01:
> Sadly, Chrome OS won't work with Nvidia GPU
> But if you also have AMD GPU (dGPU) then there are chances that blacklisting Nvidia GPU will let you boot.
> Try zork because it is made for ryzen cpus

thanks @ameyjoshi3068 but as I said:  tried to install Zork, but at this page ( ChromeOS rootfs... ) the page stops and does not change.

mesmaili151, 2021-05-17:
any update about install on amd cpu and nvidia graphics?