sebanc/brunch #837: Trackpad buttons don't work

tungmeister, 2021-01-29:
I've just updated to the lastest brunch-testing and beta 89 and my trackpad buttons aren't working with my alps i2c trackpad on my Dell Latitude 5480 - I can click by tapping the trackpad but the buttons don't function

tungmeister, 2021-01-30:
To add the set of buttons below the keyboard still work - as does the touchpoint it's just the buttons below the trackpad that aren't functional. I've tried the alt touchpad config options with no change.

suoko, 2021-03-15:
The same on 3340
What is the best option in grub for you @tungmeister  ?

milletnis, 2021-04-05:
same problem for me with my Latitude E7470
seems to be related to this driver issue here:
https://askubuntu.com/questions/1271281/dell-precision-7550-physical-mouse-buttons-behaving-like-clickpad