sebanc/brunch #838: Black screen on today r88 Acer C720

mikka9, 2021-01-31:
Black screen (only mouse cursor) at  google login phase.
r87 is OK

sebanc, 2021-01-31:
Could you try to add "i915.enable_dpcd_backlight=1" on the kernel command line ?

mikka9, 2021-01-31:
no effect
https://photos.app.goo.gl/GUwuopxd5iXsQQhH9

mikka9, 2021-01-31:
No effect  "i915.enable_dpcd_backlight=1"
Fixed after  "enable_dpcd_backlight=1"
Is this correct?

mikka9, 2021-01-31:
Tried
No effect

sebanc, 2021-01-31:
I had a look at the kernel source to see what could cause that but for now it is difficult to say, could you try the below kernel command line parameters (one by one):
- i915.enable_dpcd_backlight=0
- i915.fastboot=1
- i915.fastboot=0

mikka9, 2021-01-31:
Somewhat happens at the screen (blinking, switching...) but not solving the issue
If you want a video of system loading, I provide you, but
I stay at r87

sebanc, 2021-02-01:
A potential fix was included in the latest brunch unstable if you want to try it

mikka9, 2021-02-02:
First run was a fail
Second seemed to be ok...but no
[image]
[image]
Thank you for your job

mikka9, 2021-02-04:
I've tried all  4 combination of i915.enable_dpcd_backligh and i915.fastboot with test brunch 20210201 also.
There are graphic problems/black streen/stuck.
While brunch r87 works fine

mikka9, 2021-02-04:
I have tested with 5.10 kernel also.
What is the upgrades form r87 to r88, so fatal to C720 notebook

sebanc, 2021-02-04:
Thanks for the tests, could you confirm the below test cases as it would help me to identify the issue ?

- Brunch r87 stable 20201227 + kernel 4.19 => works
- Brunch r87 stable 20201227 + kernel 5.4 => works
- Brunch r87 stable 20201227 + kernel 5.10 => works

- Brunch r88 stable 20210131 + kernel 4.19 => works
- Brunch r88 stable 20210131 + kernel 5.4 => does not work
- Brunch r88 stable 20210131 + kernel 5.10 => does not work

mikka9, 2021-02-04:
Brunch r87 stable 20201227 + default kernel => works
Brunch r87 stable 20201227 + kernel 4.19 => works
Brunch r87 stable 20201227 + kernel 5.10 => does not work

Now I 'm using Brunch r88 unstable 20210201
+ default kernel => does not work 
 + kernel 4.19 => works (coudn't login)
 + kernel 5.1 => does not work

Configm.

mikka9, 2021-02-04:
I've created new img files, so

r87+default kernel works ok
r88+4.19 kernel works ok

Using last one by default
Thank you)

sebanc, 2021-02-07:
Could you try the build in the "brunch-unstable" repository ?

mikka9, 2021-02-07:
New generated rammus87+unstable0207
default menuitem
First boot: freezed on lang selecting (first screen)
Second boot: successfully finished logging in as new user, freezed on loggin in
Third boot: freezed after loggin in, no desktop 
"Freezed" means black screen, only mouse pointer. no responce

mikka9, 2021-02-07:
kernel-4.19 seemed to be working