sebanc/brunch #839: Tablet Mode Detection only @ 360 degrees screen rotation. Thinkpad Yoga 12

sumcda, 2021-01-31:
My Thinkpad Yoga 12 with Rammus and unstable Brunch is having an issue regarding Tablet Mode. 

On this Thinkpad Yoga 12, tablet mode should detect and switch automatically once the display is rotated passed 180 degree (might be 270 degrees). At the current time, tablet mode only switches once the display it rotated  the complete 360. 

When I first installed Chrome OS using stable brunch it had been working. I then changed the channel to beta and then developer channel and only then noticed it no longer worked. I have tried powerwash, reinstall with unstable Brunch, stable channels, different recovery images etc. 

Any ideas?

sebanc, 2021-01-31:
Could you post an "lsmod" ? (run "lsmod > ~/Downloads/lsmod.txt" in crosh shell and attach lsmod.txt from your downloads folder)

Also, could you try to add "module_blacklist=intel_hid" on the kernel command line ?

sumcda, 2021-01-31:
[lsmod.txt](https://github.com/sebanc/brunch/files/5899874/lsmod.txt)
[dmesg.txt](https://github.com/sebanc/brunch/files/5899875/dmesg.txt)

Thanks for looking into this for me. I'll have to look into how to edit kernel command line. I am by no means a expert. I have added the dmesg.txt as I saw that you asked someone with a similar issue for this output also.

sumcda, 2021-01-31:
I restarted Thinkpad, pressed 'e' at the Chrome OS screen on startup. Added module_blacklist=intel_hid to one of the lines and then pressed ctrl-x. Was this correct? 

Upon restart I rotated screen. No effect on tablet mode until full 360.

sumcda, 2021-01-31:
Line 906 dmesg.txt

[   64.281634] thinkpad_acpi: Tablet mode switch found (type: GMMS), currently in laptop mode

sebanc, 2021-01-31:
Could you try the same with "module_blacklist=intel_vbtn" (add it after cros_debug" in the grub config"

sumcda, 2021-01-31:
Tried module_blacklist=intel_vbtn. No change.

sebanc, 2021-01-31:
Could you try "module_blacklist=thinkpad_acpi" ?

sumcda, 2021-01-31:
Tried "module_blacklist=thinkpad_acpi"

Now 'Tablet Mode' does not enable even @ 360 degree rotation.

sebanc, 2021-01-31:
What's the output of "cat /sys/class/dmi/id/chassis_type" in crosh shell ?

sumcda, 2021-01-31:
10

sebanc, 2021-01-31:
What's the output of "ls /sys/bus/acpi/devices/" ?

sumcda, 2021-01-31:
ACPI0003:00  device:09  device:14  device:1f  device:2a  device:35  device:40  device:4b  device:56   LEN200B:00   LNXPOWER:02  PNP0000:00  PNP0C02:01  PNP0C0F:03
ACPI000C:00  device:0a  device:15  device:20  device:2b  device:36  device:41  device:4c  device:57   LNXCPU:00    LNXPOWER:03  PNP0100:00  PNP0C02:02  PNP0C0F:04
device:00    device:0b  device:16  device:21  device:2c  device:37  device:42  device:4d  device:58   LNXCPU:01    LNXPOWER:04  PNP0103:00  PNP0C04:00  PNP0C0F:05
device:01    device:0c  device:17  device:22  device:2d  device:38  device:43  device:4e  device:59   LNXCPU:02    LNXPWRBN:00  PNP0200:00  PNP0C09:00  PNP0C0F:06
device:02    device:0d  device:18  device:23  device:2e  device:39  device:44  device:4f  device:5a   LNXCPU:03    LNXSYBUS:00  PNP0501:00  PNP0C09:01  PNP0C0F:07
device:03    device:0e  device:19  device:24  device:2f  device:3a  device:45  device:50  INT0800:00  LNXCPU:04    LNXSYBUS:01  PNP0501:01  PNP0C0A:00  PNP0C14:00
device:04    device:0f  device:1a  device:25  device:30  device:3b  device:46  device:51  INT33D6:00  LNXCPU:05    LNXSYSTM:00  PNP0A05:00  PNP0C0C:00  PNP0C14:01
device:05    device:10  device:1b  device:26  device:31  device:3c  device:47  device:52  INT340E:00  LNXCPU:06    LNXVIDEO:00  PNP0A08:00  PNP0C0D:00  SMO1200:00
device:06    device:11  device:1c  device:27  device:32  device:3d  device:48  device:53  INT340F:00  LNXCPU:07    MSFT0001:00  PNP0B00:00  PNP0C0F:00
device:07    device:12  device:1d  device:28  device:33  device:3e  device:49  device:54  INT3F0D:00  LNXPOWER:00  MSFT9000:00  PNP0C01:00  PNP0C0F:01
device:08    device:13  device:1e  device:29  device:34  device:3f  device:4a  device:55  LEN0068:00  LNXPOWER:01  MSFT9001:00  PNP0C02:00  PNP0C0F:02

sebanc, 2021-01-31:
Did you install a firmware update at some point ?
From what I can see:
- "intel_vbtn" does not send tablet mode events on your device
- which makes sense as "thinkpad_acpi" module handle it

There was no significat change to those modules recently so I do not see how it could have worked unless it was broken by a firmware update.

sumcda, 2021-01-31:
I was using Linux Mint and decided to install your Brunch Chrome OS. It was working in tablet mode perfectly. I first noticed it was not working when the backlit keyboard lights stayed on when in 'tablet mode' position. I am not sure what caused the change but I had been experimenting with changing the channel to developer etc. I also enable linux beta.

I wouldn't know how to change the firmware without an official windows update/Lenovo update. Would there be another way?

sebanc, 2021-01-31:
It will be difficult to debug this unless you can reproduce the situation where it was working using older builds.

Right now the only thing I can think of is resetting chrome sync first and then perform a fresh install. It should give you a completely stock chromeos.

To reset chrome synced data:
- Login to your Google dashboard account
- Go to the "Chrome" application -> Manage synced data
- Click on "reinitialize synced data"

sumcda, 2021-01-31:
I will give that a try. Could resetting BIOS settings help?

sebanc, 2021-01-31:
Yes that's a good idea

sumcda, 2021-01-31:
It worked! I reset BIOS setting to device recommended and it seems to be working.

sebanc, 2021-01-31:
Nice to see that you fixed it :)

sumcda, 2021-01-31:
Thank you for your help. I have another different issue but will post a new thread when I get the chance.

librehat, 2021-02-02:
I'm using a different laptop and I noticed that the tablet mode works similarly to what you described. If I rotate it completely 360 degrees, the keyboard will no longer be backlit nor responding to keystrokes. However, my Chrome OS doesn't change into tablet mode.

Did you have to add those modules blacklists at all (besides resetting BIOS)?

sumcda, 2021-02-02:
I can confirm that all I did to fix my issue was to enter BIOS and reset to recommended settings. When I restarted, tablet mode worked and continues to work. I have updated to OS 88 too with no problems.  

The backlists made no difference.

Have you tried Shift-Ctrl-Alt-T to force tablet mode? For me it forces tablet mode but does not deactivate keyboard or backlight and there is no virtual keyboard. It may act differently for you. If you have specific questions about my current reset BIOS settings let me know and I'll take a look.

librehat, 2021-02-03:
@sumcda How did you update to OS 88 btw? I can't find the recovery image for CrOS 88.

So for me, if I force it into tablet mode, it doesn't activate the virtual keyboard. However, it does deactivate my physical keyboard (but not the touchpad). I think it's a different issue for my model and I'll try to update to the latest CrOS 88 and the brunch framework and then check again (and open a new thread so this doesn't become irrelevant here)

sumcda, 2021-02-03:
To update to OS 88 to need to enter the grub config by pressing 'e' on the keyboard when you see the Chrome OS text upon startup. On one of the lines you will see the text "cros_debug". Add "options=enable_updates" after it and save the changes. When the computer has restarted you can then go to the settings and 'search for updates'. OS 88 will be automatically downloaded and updated. This is the same method to change the OS channel to 'Beta' or 'Developer' which will let you try OS 89 and OS 90.

As for your issue with the 'tablet mode', I agree it is probably best for you to start a new thread. It may be something trivial. Good luck.

closed 2026-03-18 by github-actions[bot]