sebanc/brunch #840: ANDROID APPS ARE TAKING TIME TO LOAD .

thehalfnotes, 2021-01-31:
android apps are taking time to load and sometimes are not opening.

b4iterdev, 2021-02-10:
Are you using HDD or SSD ? if you are using HDD then it's the problem

khanhtranngoccva, 2021-02-11:
Nothing you can do about it. Wait for 2 minutes or install Chrome OS on an SSD/NVMe.

Zarvis12, 2021-02-11:
[image]

Change this setting to enable hyperthreading you will see some improvement

https://support.google.com/chromebook/answer/9340236?hl=en

khanhtranngoccva, 2021-02-11:
That doesn't work, I have tried this option ages ago. Just grab an SSD

b4iterdev, 2021-02-11:
HyperThreading only improve task processing, instead of enable it on chome://flags. just use enforce_hyperthreading=1 kenel parameter.