sebanc/brunch #841: Grub error on BIOS

supersri811, 2021-02-01:
error: no such partition
You need to boot the kernel first
This happens even after I extract the mbr_support.tar.gz package

supersri811, 2021-02-01:
I have an i3 processor and the computer is hp pavilion dv6

sebanc, 2021-02-01:
This usually means that rootc.img from the brunch archive has not been written to partition 7, did you unpack it from the brunch archive ?

supersri811, 2021-02-01:
yes

sebanc, 2021-02-01:
Are you trying to dual boot or to install to hdd ?

supersri811, 2021-02-01:
dual boot

supersri811, 2021-02-01:
actually triple boot , windows, linux mint and chrome os

sebanc, 2021-02-01:
As written in the main readme, dual boot is not supported with mbr because most legacy grub do not support it.

However, there were some success reports with the legacy grub from arch linux.

supersri811, 2021-02-01:
Okay I'll try that

closed 2021-02-01 by supersri811