sebanc/brunch #842: Lenovo Thinkpad Yoga 11e 6th gen issues

Lucaacer, 2021-02-02:
I have just bought a Lenovo Thinkpad Yoga 11e 6th gen with a core m3-8110y and 8 gigs ram. More or less the config of the most powerful Surface Go.

What does not work: crostini and android apps.

What works: everything else, including rotation, tablet switch, the stylus and so on.

I tested rammus, samus and nami builds with the latest brunch framework.

Crostini doesn't give any sign of life. After the first boot, typing in crosh terminal vmc start termina produced an output about terminal and parallel not being enabled (I don't exatcly remember it). Now the command is unknown.

The android container works after the very first reboot only and the android init fix does not seem to work in my case.

It's a pity, because performance is very good even when running the OS off an usb key.

Btw, this device has got 2 cameras and they work both after removing /etc/camera/

Thanks as usual.

sebanc, 2021-02-04:
For android apps,try to reset Chrome sync:
Login to your Google dashboard account -> Go to the "Chrome" application -> Manage synced data and click on "reinitialize synced data"

For crostini, is virtualisation enabled in your bios ?

Lucaacer, 2021-02-04:
Virtualisation was actually disabled, I hadn't checked It because it was enabled in Myanmar former 11e 5th gen. Sorry.

Android, in the other side, still doesn't work, even if I supposedly  performed the steps you suggested.

Lucaacer, 2021-03-07:
I haven't been able to run android apps, so far.

This is the log with the latest brunch framework and rammus build. Usb key or sdd don't change the game 

chronos@localhost / $ cat /var/log/messages | grep Android
2021-03-07T18:58:54.539664+01:00 INFO session_manager[1598]: [INFO:session_manager_impl.cc(1872)] Android container with PID 3111 stopped
2021-03-07T19:00:08.285973+01:00 ERR session_manager[1598]: [ERROR:android_oci_wrapper.cc(156)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2021-03-07T19:00:13.404669+01:00 ERR session_manager[1598]: [ERROR:session_manager_impl.cc(1779)] Starting Android container failed.
2021-03-07T19:00:13.404740+01:00 ERR session_manager[1598]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.

If you have got any idea, I'll be glad to test it or give you more info.

Thanks:)

Lucaacer, 2021-03-26:
Well, what I've just found out it's very weird: the reason why the android container does not start is actually tied to /etc/camera/

I mean, in order to let my camera work, I need to remove such folder but, when the device reboots, the container does not start.

If I just remove camera_characteristics.conf or I  make a void .conf file without deleting the folder,  the .conf file gets rewritten every time I reboot my laptop.

The only workaround it this one:

ctrl + alt + t

type "shell"

type "cd /etc/camera/"
type "sudo mv camera_characteristics.conf camera_characteristics.conf.bak"

Open another tab and go to chrome://flags

type "reset all"

type "restart"

so that the UI will restart without replacing camera_characteristics.conf and with the android container perfectly working.

Very very strange, I frankly don't get why messing with /etc/camera/ should block the android container, but I tested such issue five or six times straight and it actually does it.

I hope that sharing my findings might be useful for any of you;)

sebanc, 2021-03-27:
Could you try a clean install of the latest brunch-unstable with "options=no_camera_config" added in the grub configuration ?
https://github.com/sebanc/brunch-unstable/releases/

Lucaacer, 2021-03-27:
> Could you try a clean install of the latest brunch-unstable with "options=no_camera_config" added in the grub configuration ?
> https://github.com/sebanc/brunch-unstable/releases/

Thanks, it actually worked;)

I was already testing the latest unstable release and editing the grub works.

On the other side, I had to configure a custom crostini container through crosh, since the "normal" graphical installation did not work, maybe due to my almost worn out small usb key.

Anyway, everything is ok and I suppose you might include this grub option to the stable release,

Lucaacer, 2021-03-28:
Just to report that everything works after installing the OS on my ssd in dual boot with windows pro.

Btw, the custom code you have got to insert in grub2win does not require neither the first brace "{" nor the final one "}". 

I mean, if you don't remove those braces,  chrome OS will never boot. 

Braces are  required, on the other side, when you add an entry to an actual linux grub.

Maybe you might pin such tips somewhere:)

closed 2026-03-18 by github-actions[bot]