sebanc/brunch #843: i am facing a issue in which it asy loading brunch framworks .... and restarts i waited for a few hours but there was no diffrece

Clash467, 2021-02-02:


Clash467, 2021-02-02:
sudo sh install.sh
Ign:1 cdrom://Linux Mint 20.1 _Ulyssa_ - Release amd64 20210103 focal InRelease
Err:2 cdrom://Linux Mint 20.1 _Ulyssa_ - Release amd64 20210103 focal Release
  Please use apt-cdrom to make this CD-ROM recognized by APT. apt-get update cannot be used to add new CD-ROMs
Hit:3 http://archive.ubuntu.com/ubuntu focal InRelease
Hit:4 http://archive.ubuntu.com/ubuntu focal-updates InRelease                 
Hit:5 http://security.ubuntu.com/ubuntu focal-security InRelease               
Hit:6 http://archive.canonical.com/ubuntu focal InRelease                      
Hit:7 http://archive.ubuntu.com/ubuntu focal-backports InRelease               
Ign:8 http://packages.linuxmint.com ulyssa InRelease                          
Hit:9 http://packages.linuxmint.com ulyssa Release
Reading package lists... Done
E: The repository 'cdrom://Linux Mint 20.1 _Ulyssa_ - Release amd64 20210103 focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
figlet is already the newest version (2.2.5-3).
0 upgraded, 0 newly installed, 0 to remove and 132 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
pv is already the newest version (1.6.6-1).
0 upgraded, 0 newly installed, 0 to remove and 132 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cgpt is already the newest version (0~R81-12871.B-1).
0 upgraded, 0 newly installed, 0 to remove and 132 not upgraded.
        ____  _   _ ____ ____   ____ ____  ___ ____  _____   _____ ___  
       / ___|| | | | __ ) ___| / ___|  _ \|_ _| __ )| ____| |_   _/ _ \ 
       \___ \| | | |  _ \___ \| |   | |_) || ||  _ \|  _|     | || | | |
        ___) | |_| | |_) |__) | |___|  _ < | || |_) | |___    | || |_| |
       |____/ \___/|____/____/ \____|_| \_\___|____/|_____|   |_| \___/ 
                                                                        
                           _  __        _            
                          | |/ /___  __| | __ _ _ __ 
                          | ' // _ \/ _` |/ _` | '__|
                          | . \  __/ (_| | (_| | |   
                          |_|\_\___|\__,_|\__,_|_|   
                                                     
                 _   _ _           _           _ _              
                | \ | (_)_ __ ___ | |__   __ _| | | ____ _ _ __ 
                |  \| | | '_ ` _ \| '_ \ / _` | | |/ / _` | '__|
                | |\  | | | | | | | |_) | (_| | |   < (_| | |   
                |_| \_|_|_| |_| |_|_.__/ \__,_|_|_|\_\__,_|_|   
                                                                
https://www.youtube.com/user/kedar123456889
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
ERROR: GptSanityCheck() returned 2: GPT_ERROR_INVALID_HEADERS
chromeos-install.sh: line 187: [: -eq: unary operator expected
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
ERROR: GptSanityCheck() returned 2: GPT_ERROR_INVALID_HEADERS
chromeos-install.sh: line 187: [: -gt: unary operator expected
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
ERROR: GptSanityCheck() returned 2: GPT_ERROR_INVALID_HEADERS
All data on device /dev/sda will be lost, are you sure ? (type yes to continue) yes
umount: /dev/sda: not mounted.
umount: /dev/sda1: not mounted.
umount: /dev/sda10: not mounted.
umount: /dev/sda11: not mounted.
umount: /dev/sda12: not mounted.
umount: /dev/sda2: not mounted.
umount: /dev/sda3: not mounted.
umount: /dev/sda4: not mounted.
umount: /dev/sda5: not mounted.
[... 489 lines cut ...]
Command 'pv:' not found, did you mean:

  command 'pvs' from deb lvm2 (2.03.07-1ubuntu1)
  command 'pv' from deb pv (1.6.6-1)
  command 'pvm' from deb pvm (3.4.6-2build2)

Try: sudo apt install <deb name>

mint@mint:/cdrom/Chrome OS$ Writing partition 11
Writing: command not found
mint@mint:/cdrom/Chrome OS$ pv: /dev/loop1p11: No such file or directory

Command 'pv:' not found, did you mean:

  command 'pv' from deb pv (1.6.6-1)
  command 'pvs' from deb lvm2 (2.03.07-1ubuntu1)
  command 'pvm' from deb pvm (3.4.6-2build2)

Try: sudo apt install <deb name>

mint@mint:/cdrom/Chrome OS$ Writing partition 12
Writing: command not found
mint@mint:/cdrom/Chrome OS$ 32.0MiB 0:00:01 [22.7MiB/s] [================================>] 100%            
bash: ]: Read-only file system
mint@mint:/cdrom/Chrome OS$ ChromeOS installed.

Clash467, 2021-02-02:
why is it

Clash467, 2021-02-02:
@sebanc pls help

Clash467, 2021-02-02:
sudo sh install.sh  
Ign:1 cdrom://Linux Mint 20.1 _Ulyssa_ - Release amd64 20210103 focal InRelease
Err:2 cdrom://Linux Mint 20.1 _Ulyssa_ - Release amd64 20210103 focal Release
  Please use apt-cdrom to make this CD-ROM recognized by APT. apt-get update cannot be used to add new CD-ROMs
Hit:3 http://archive.ubuntu.com/ubuntu focal InRelease
Hit:4 http://archive.ubuntu.com/ubuntu focal-updates InRelease                 
Hit:5 http://security.ubuntu.com/ubuntu focal-security InRelease               
Hit:6 http://archive.ubuntu.com/ubuntu focal-backports InRelease               
Hit:7 http://archive.canonical.com/ubuntu focal InRelease                      
Ign:8 http://packages.linuxmint.com ulyssa InRelease                          
Hit:9 http://packages.linuxmint.com ulyssa Release
Reading package lists... Done
E: The repository 'cdrom://Linux Mint 20.1 _Ulyssa_ - Release amd64 20210103 focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
figlet is already the newest version (2.2.5-3).
0 upgraded, 0 newly installed, 0 to remove and 132 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
pv is already the newest version (1.6.6-1).
0 upgraded, 0 newly installed, 0 to remove and 132 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cgpt is already the newest version (0~R81-12871.B-1).
0 upgraded, 0 newly installed, 0 to remove and 132 not upgraded.
        ____  _   _ ____ ____   ____ ____  ___ ____  _____   _____ ___  
       / ___|| | | | __ ) ___| / ___|  _ \|_ _| __ )| ____| |_   _/ _ \ 
       \___ \| | | |  _ \___ \| |   | |_) || ||  _ \|  _|     | || | | |
        ___) | |_| | |_) |__) | |___|  _ < | || |_) | |___    | || |_| |
       |____/ \___/|____/____/ \____|_| \_\___|____/|_____|   |_| \___/ 
                                                                        
                           _  __        _            
                          | |/ /___  __| | __ _ _ __ 
                          | ' // _ \/ _` |/ _` | '__|
                          | . \  __/ (_| | (_| | |   
                          |_|\_\___|\__,_|\__,_|_|   
                                                     
                 _   _ _           _           _ _              
                | \ | (_)_ __ ___ | |__   __ _| | | ____ _ _ __ 
                |  \| | | '_ ` _ \| '_ \ / _` | | |/ / _` | '__|
                | |\  | | | | | | | |_) | (_| | |   < (_| | |   
                |_| \_|_|_| |_| |_|_.__/ \__,_|_|_|\_\__,_|_|   
                                                                
https://www.youtube.com/user/kedar123456889
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
ERROR: GptSanityCheck() returned 2: GPT_ERROR_INVALID_HEADERS
chromeos-install.sh: line 187: [: -eq: unary operator expected
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
ERROR: GptSanityCheck() returned 2: GPT_ERROR_INVALID_HEADERS
chromeos-install.sh: line 187: [: -gt: unary operator expected
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
ERROR: GptSanityCheck() returned 2: GPT_ERROR_INVALID_HEADERS
All data on device /dev/sda will be lost, are you sure ? (type yes to continue) yes
umount: /dev/sda: not mounted.
umount: /dev/sda1: not mounted.
umount: /dev/sda10: not mounted.
umount: /dev/sda11: not mounted.
umount: /dev/sda12: not mounted.
umount: /dev/sda2: not mounted.
umount: /dev/sda3: not mounted.
sumount: /dev/sda4: not mounted.
umount: /dev/sda5: not mounted.
[... 197 lines cut ...]
pv: /dev/loop1p1: No such file or directory
Writing partition 2
pv: /dev/loop1p4: No such file or directory
Writing partition 3
pv: /dev/loop1p3: No such file or directory
Writing partition 4
pv: /dev/loop1p4: No such file or directory
Writing partition 5
pv: /dev/loop1p3: No such file or directory
Writing partition 6
pv: /dev/loop1p6: No such file or directory
Writing partition 7
1.00GiB 0:00:43 [23.8MiB/s] [================================>] 100%            
Writing partition 8
pv: /dev/loop1p8: No such file or directory
Writing partition 9
pv: /dev/loop1p9: No such file or directory
Writing partition 10
pv: /dev/loop1p10: No such file or directory
Writing partition 11
pv: /dev/loop1p11: No such file or directory
Writing partition 12
32.0MiB 0:00:01 [22.4MiB/s] [================================>] 100%            
ChromeOS installed.
 installed.

ameyjoshi3068, 2021-02-02:
It looks like you need help for installing Chrome OS. I can help you in this case.
Please tell me your pc specifications(CPU and Graphics card) to tell whether your pc can run Chrome OS or not.

thehalfnotes, 2021-02-03:
Is it when you boot your chrome os (brunch) it boot then restarts and boot again-again, if yes then you can go to the kernel command line and add any option and boot the command line ..  and YEAH it works... I faced it today

ameyjoshi3068, 2021-02-03:
As far as I understand the main problem can be related to the Linux mint and not brunch/Chrome OS. So one solution can be installing Linux Mint on a PC. 
I would recommend you to follow the setup procedure mentioned in Readme, not on any youtube video. If you didn't understand any part of the setup then I am here to help you.

ameyjoshi3068, 2021-02-03:
@thehalfnotes  can you please tell me what exactly you had done to solve the reboot problem?

Clash467, 2021-02-03:
i have intel i7 m640 with nvidia nvs m3100 pls help me

Clash467, 2021-02-03:
what should i add in the cmd line 
pls tell i am in =need 
@thehalfnotes @ameyjoshi3068

thehalfnotes, 2021-02-03:
> @thehalfnotes can you please tell me what exactly you had done to solve the reboot problem?
> I have the same problem with the default kernel.

@Clash467, when you press boot key/ power key of the machine then immediately press "e" key and kernel command line will be open then go to the last of first cros_debug and give a space and type options=android_init_fix  *any options here which are suitable for your machine try this a while * then press ctrl+x and let it boot .... maybe hip hip ..YaY

ameyjoshi3068, 2021-02-03:
@Clash467 Nvidia graphics card are not supported by Chrome OS. You need intel graphics.
There are chances that your PC may have inbuilt Intel graphics as another GPU.
You can try adding `module_blacklist=nouveau` after cros_debug and before options=

ameyjoshi3068, 2021-02-03:
@thehalfnotes  Due to faulty installation by Clash 467 he doesn't have complete chromeos.img so the method we discussed above will not gonna work

ameyjoshi3068, 2021-02-03:
The script written by Kedar Nimbalker is not 100% correct in all circumstances. So he will need help for the correct installation of Chrome OS.

ameyjoshi3068, 2021-02-03:
@Clash467 You can try the installation of Chrome OS anyway But it can fail to boot due to your graphics card. For that,

1. Install Linux mint on your PC with the help of your USB.(Search for "install Linux Mint" on the desktop.)
2. After installation of Linux Mint, Open terminal from the bottom shelf and Run below commands one by one
     `sudo apt update` and then enter password
     `sudo apt install pv tar cgpt`
3. You should have those 6 files in Chrome OS folder(as mentioned in video) Now open in terminal.
4. Type below command
     `sudo bash chromeos-install.sh -src rammus.bin -dst /dev/sda`

Post here the output of 4. step

Don't hesitate to comment if you have any problems.

Clash467, 2021-02-03:
> Is it when you boot your chrome os (brunch) it boot then restarts and boot again-again, if yes then you can go to the kernel command line and add any option and boot the command line .. and YEAH it works... I faced it today

How do u do it???

thehalfnotes, 2021-02-03:
@Clash467  @ameyjoshi3068  I installed  chrome OS using brunch... Using kedar Nimbalkar's tutorial video on a Machine running on AMD cou... And ... Working well... . In the morning I was editing the grub options and my machine's battery was going out of charge... And the result machine got sudden Shutdown during the brunch option updating..  When I plugged it and boot it again as result  , a brunch boot loop... I went to the command line * when you boot ** power on your machine using key * * your machine laptop/pc press ' e ' the command line will appear and go to the end of cros_debug and any  ' option ' such as cros_debug options=android_init_fix and then press Ctrl+x and let it boot... Now it not should be in boot loop.

ameyjoshi3068, 2021-02-04:
@thehalfnotes  I understood your case. It is completely different than me. 
Nice to see you fixed corrupt rootfs.
@Clash467 The technique @thehalfnotes said will not work for you.