sebanc/brunch #844: [Compatibility feedback] Incompatible with 'scarlet' recovery image

SemiZhang, 2021-02-02:
Hi, I've tried to create a brunched disk image with a 'scarlet' image, but the installation script refused to do such and return an error message "chromeos_13505.111.0_scarlet.bin is not a valid ChromeOS recovey image (have you unzipped it ?)"
The same procedure worked well with 'rammus' image.

I know that not all image are compatible for the moment, this is simply a feedback.

Besides, the 'unibulid images' has been mentioned in ReadMe document, but i have found any information about where to find one. I will be appreciated if anyone could tell me something about it.
'unibuild images: intended to manage multiple devices through the use of the CrosConfig tool.'

sebanc, 2021-02-02:
I think that now all recovery images are unibuild, previously some were not.

scarlet is an arm architecture recovery image so it is not compatible with brunch which only supports x86_64.

SemiZhang, 2021-02-02:
> I think that now all recovery images are unibuild, previously some were not.
> 
> scarlet is an arm architecture recovery image so it is not compatible with brunch which only supports x86_64.

Thank you so much for your information _sebanc_.
It's weird that, according to 'cros-updates-serving.appspot.com', 'scarlet' has been used by 'Acer Chromebook Tab 10' which is equipped with 'Intel Atom X5-Z8300' chipset (supposed to be x86). Maybe I got wrong information about it.

Otherwise, for those who is trying to deploy ChromeOS on '**Intel Atom Cherry Trail**' chipset, the image '**banon**' has significantly a better performance. (x5-Z8350 with Lenovo miix320 in my case) 

p.s. On my Lenovo miix320, the brightness of screen is broken with kernel 5.4 (unconfigurable, flicking after waking up), but works well on both 4.19 and 5.10.

angelvel1625, 2021-02-03:
hey i tried to download the most recent version of brunch and in the pag of downloads say "error:bloqued"

closed 2022-05-18 by SemiZhang