sebanc/brunch #846: Teclast F6 plus Wifi

wilyamsoys, 2021-02-03:
Hello! 
I just installed Chrome OS on my Teclast F6 Pro using Rammus 87 and brunch r88, dual boot with windows 10.
From what I can see, all works ok with /kernel-5.10, except for WiFi 

The device uses intel AC 3615 WLAN card, and works ok in Windows.
I already added options=iwlwifi_backport in the grub configs file, but it doesn't seem to work. I'm now stuck at the "conmect a network" prompt.. and going to the taskbar>Network, though I can toggle wi-fi on using the blue switch, the line below it still says "Wi-Fi is turned off"

wilyamsoys, 2021-02-03:
Ran sudo rfkill list all and this is what I get.

sebanc, 2021-02-04:
Could you try 5.10 kernel without "options=iwlwifi_backport" ? (the iwlwifi driver in the 5.10 kernel is more recent than the iwlwifi_backport one)

If it still does not work, could you post the output of "dmesg | grep iwlwifi" output in the console (CTRL+ALT+F2 and login as "root") ?

MadhiasM, 2021-05-17:
I am using the Teclast F6 **Pro** which uses the Intel AC 3165 WLAN card aswell. On Kernel 5.4 it works just fine, so if there is no need for kernel 5.10 on your side you could switch back to kernel 5.4 I guess. Or is there anything else that only works with kernel 5.10?