sebanc/brunch #850: Lenovo Miix 320 - camera and bluetooth don't work

Rytuklis, 2021-02-03:
Hi there,

I'm using Brunch r88 stable 20210131 with Rammus recovery 87. It's a 2 in 1 Lenovo Miix 320 hybrid tablet. Everything seems to work fine, except for camera and bluetooth. Any possible solutions?

sebanc, 2021-02-04:
Hi,

Unfortunately the cameras seem to be ipu3 so they are not currently supported by the linux kernel.

Regarding bluetooth, you cannot pair phones in chromeos, can you pair audio (headphones, speakers) or input devices ?

Rytuklis, 2021-02-04:
Hi,

Thank you for your reply, Sebanc.

I thought that Bluetooth wasn't working, since when browsing Bluetooth device you'd always see a bunch of phones and stuff. However, it detected my BT headphones just fine. So that works perfectly finel.

However, as for the cameras, they did seem to work with other Linux distributions, such as Pop OS and Linux Mint, so is it really a Linux Kernel problem?

Kind regards,
Rytis

sebanc, 2021-02-04:
For the cameras, what's the output of `dmesg | grep uvcvideo` in crosh shell ?

Regarding sensors, your device does not use the common intel sensor hub which is the only driver that has been ported for chromeos (it uses "axp288" apparently). Someone with the right hardware would need to port this driver for it to work in chromeos.

closed 2026-03-18 by github-actions[bot]