sebanc/brunch #851: Kernel 5.10 doesn't work properly with pixel 2013

dreit-p, 2021-02-04:
When I updated from 85 to 88 I noticed everything lags, but the diagnostic app doesn't show high loading of CPU. I checked coolers and heard it starts and stops immediately every 2 seconds. I found [ectool](https://github.com/MrChromebox/firmware/issues/115#issuecomment-429084785) and run  `sudo ./ectool autofanctrl` and it fixed fans till restarting. Everything was still laggy even with working fans.
Then I decided to use the same kernel that you used with 85 (it is 4.19) and everything is great. My notebook is fast again and fans are working properly.
I guess kernel 5.10 gives some advantages and maybe you can check config? I'll try to help as much as I can. :)

sebanc, 2021-02-04:
Kernel 5.10 is experimental for now and indeed suffer from a i/o lag issue (which is present in mainline kernel 5.10 as well and still not completely fixed in kernel 5.11). For now I would suggest to use kernel 5.4, why are you using kernel 4.19 ?

dreit-p, 2021-02-04:
> why are you using kernel 4.19 ?

I just set the last kernel that worked fine 😁

sebanc, 2021-02-04:
Ok so your fans issue is with kernel 5.4, right ?

Is lag also an issue with kernel 5.4 ?

dreit-p, 2021-02-06:
Yes, the issue with default 5.4 kernel. It was lagging when it was without changing of the grub config yet. Now I use 4.19 and it everything is okay.

sebanc, 2021-02-07:
Could you try the build in the "brunch-unstable" repository ?

dreit-p, 2021-02-15:
@sebanc 
Sorry, I cannot afford to allow the data wiping that will happen due to system reinstallation yet. Maybe later, but now the laptop is being actively used. Maybe I can somehow test the changes in a different way?

dreit-p, 2021-02-24:
@sebanc I tried your [the latest release](https://github.com/sebanc/brunch/releases/tag/r88-stable-20210223) and this didn't fix it