sebanc/brunch #853: Surface 3 (Non-Pro) working great but backlight stays on when locked

LilChippyD, 2021-02-05:
Hey Guys! Love this project and fantastic work! r87 running with Rammus and there's basically only two small issues that I've experienced and just wondering if anyone has fixed any of them or if there is some ChromeOS commands (this is my first experience with ChromeOS) I can run to fix them.

1) When locking my Surface 3, I can still see the backlight. I've left it running like this to see if it would eventually shut off but no luck. Wondering if there is a setting to actually put it to sleep but until then, I just shut it down and power it back up from off, which isn't the biggest deal.

2) Plug in headphones are not working, which is fine because my bluetooth buds are.

Basically hoping there's some fix for the backlight so I can just have it sleeping and pick it up and go.

sebanc, 2021-02-07:
For backlight, could you try the build in the "brunch-unstable" repository ?

Regarding headphones, they should work if you reboot while they are plugged.

netson99, 2021-02-15:
I have the same issue on the headphone and solved by running 2 command : 
     amixer -c1 numid=155 1
     amixer -c1 numid=177 1

and put it on a script /var/brunch/bootscripts/alsa.sh to run automatically on boot.

mast3rmindx, 2021-02-21:
@sebanc Hello I’m not the OP but I can confirm the unstable repo fixed my blacklight issue when putting the OS to sleep also brightness control. Now I have a fully functional ChromeOS on SP3 thank you!

LilChippyD, 2021-02-24:
Went back to Windows but will be trying the unstable right now! Will report back.

Thanks guys!

TheSwisscheese2013, 2021-03-13:
I tried it but it didn't make a difference with mine. Backlight still stays on and no brightness adjustment.

mast3rmindx, 2021-03-15:
> I tried it but it didn't make a difference with mine. Backlight still stays on and no brightness adjustment. 

I was using the 5.12 kernel maybe check if that makes a difference?