sebanc/brunch #855: [Wi-Fi] rtw88: timed out to flush queue

argraur, 2021-02-05:
Hardware: `HUAWEI NBLK-WAX9X/NBLK-WAX9X-PCB, BIOS 1.07 06/02/2020`
Wi-Fi adapter: `RTL8822CE`
Brunch version: `r88 stable 20210131`
CrOS version: `88.0.4324.109 (Official Build) (64-bit)`
Steps to reproduce: Open up a SSH connection, run `ping 8.8.8.8` in crosh.

Opening up SSH connection causes bug in rtw88 driver, resulting in ping spikes up to 1000-2000 ms.

Expected result:
```
crosh> ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=108 time=13.3 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=108 time=10.2 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=108 time=6.69 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=108 time=10.6 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=108 time=13.0 ms
64 bytes from 8.8.8.8: icmp_seq=6 ttl=108 time=7.39 ms
64 bytes from 8.8.8.8: icmp_seq=7 ttl=108 time=7.63 ms
64 bytes from 8.8.8.8: icmp_seq=8 ttl=108 time=15.1 ms
64 bytes from 8.8.8.8: icmp_seq=9 ttl=108 time=11.9 ms
64 bytes from 8.8.8.8: icmp_seq=10 ttl=108 time=8.30 ms
```

Actual result:
```
crosh> ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=108 time=117 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=108 time=756 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=108 time=752 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=108 time=830 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=108 time=77.6 ms
64 bytes from 8.8.8.8: icmp_seq=6 ttl=108 time=684 ms
64 bytes from 8.8.8.8: icmp_seq=7 ttl=108 time=2041 ms
64 bytes from 8.8.8.8: icmp_seq=8 ttl=108 time=1027 ms
64 bytes from 8.8.8.8: icmp_seq=9 ttl=108 time=10.7 ms
64 bytes from 8.8.8.8: icmp_seq=10 ttl=108 time=13.5 ms
64 bytes from 8.8.8.8: icmp_seq=11 ttl=108 time=9.62 ms
64 bytes from 8.8.8.8: icmp_seq=12 ttl=108 time=484 ms
64 bytes from 8.8.8.8: icmp_seq=13 ttl=108 time=604 ms
```

dmesg bug trace
```
[  268.562136] ------------[ cut here ]------------
[  268.562141] purge skb(s) not reported by firmware
[  268.562181] WARNING: CPU: 5 PID: 0 at drivers/net/wireless/realtek/rtw88/tx.c:159 rtw_tx_report_purge_timer+0x1e/0x48 [rtw88_core]
[  268.562183] Modules linked in: vhost_vsock vmw_vsock_virtio_transport_common vsock vhost kvm_amd ccp tpm_tis tpm_tis_core wmi snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg btusb snd_hda_codec btrtl snd_hda_core btbcm rtw88_8822ce rtw88_8822c btintel snd_hwdep rtw88_pci snd_pci_acp3x rtw88_core k10temp i2c_piix4 mac80211 i915 cfg80211 video intel_gtt amdgpu gpu_sched
[  268.562208] CPU: 5 PID: 0 Comm: swapper/5 Tainted: G        W         5.4.88-brunch-sebanc #1
[  268.562211] Hardware name: HUAWEI NBLK-WAX9X/NBLK-WAX9X-PCB, BIOS 1.07 06/02/2020
[  268.562220] RIP: 0010:rtw_tx_report_purge_timer+0x1e/0x48 [rtw88_core]
[  268.562224] Code: e0 5b 5d 41 5c 41 5d 41 5e 41 5f c3 0f 1f 44 00 00 83 7f d8 00 74 3c 41 54 55 53 48 89 fb 48 c7 c7 3e 66 ad c0 e8 56 11 ed eb <0f> 0b 4c 8d 63 b0 4c 89 e7 e8 56 7c ee eb 48 8d 7b c8 48 89 c5 e8
[  268.562227] RSP: 0018:ffffa5d580280e98 EFLAGS: 00010282
[  268.562231] RAX: 0000000000000000 RBX: ffff976ce3a2a968 RCX: 0000000000000007
[  268.562233] RDX: 0000000000000006 RSI: ffffa5d580280d6c RDI: ffff976e47b56580
[  268.562235] RBP: 0000000000000101 R08: ffffffffad0e0c01 R09: ffffffffadba1fcf
[  268.562237] R10: 0000000000000000 R11: ffffa5d580280d35 R12: ffff976ce3a2a968
[  268.562239] R13: ffffffffc0ac4129 R14: ffffa5d580280ee8 R15: ffffffffc0ac4129
[  268.562242] FS:  0000000000000000(0000) GS:ffff976e47b40000(0000) knlGS:0000000000000000
[  268.562245] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  268.562247] CR2: 00001587e3372270 CR3: 00000001fb182000 CR4: 00000000003406e0
[  268.562249] Call Trace:
[  268.562254]  <IRQ>
[  268.562264]  call_timer_fn+0x8b/0x144
[  268.562269]  __run_timers+0x150/0x18e
[  268.562274]  ? hrtimer_forward+0x7b/0x87
[  268.562279]  ? timerqueue_add+0x62/0x68
[  268.562283]  ? enqueue_hrtimer+0x97/0xa1
[  268.562287]  run_timer_softirq+0x19/0x2d
[  268.562292]  __do_softirq+0x12d/0x296
[  268.562299]  irq_exit+0x46/0x8a
[  268.562304]  smp_apic_timer_interrupt+0x134/0x141
[  268.562309]  apic_timer_interrupt+0xf/0x20
[  268.562312]  </IRQ>
[  268.562318] RIP: 0010:cpuidle_enter_state+0xe7/0x19f
[  268.562322] Code: 49 89 c4 e8 08 fd ff ff 31 ff e8 d2 dd 89 ff 45 84 ff 74 12 9c 58 0f ba e0 09 73 03 0f 0b fa 31 ff e8 46 1a 8e ff fb 45 85 f6 <0f> 88 9c 00 00 00 49 63 d6 4c 89 e0 48 6b f2 68 4c 29 e8 48 6b ca
[  268.562324] RSP: 0018:ffffa5d58016fe90 EFLAGS: 00000202 ORIG_RAX: ffffffffffffff13
[  268.562327] RAX: ffff976e47b69880 RBX: ffff976e45c30800 RCX: 000000000000001f
[  268.562329] RDX: 0000000000000000 RSI: 0000000000000005 RDI: 0000000000000000
[  268.562331] RBP: ffffffffad5078e0 R08: 0000003e878ce44b R09: 000000000848bfc8
[  268.562333] R10: 071c71c71c71c71c R11: 0000000000000020 R12: 0000003e878ce44b
[  268.562335] R13: 0000003e877ddadf R14: 0000000000000001 R15: 0000000000000000
[  268.562343]  cpuidle_enter+0x2a/0x36
[  268.562348]  do_idle+0x1a9/0x213
[  268.562352]  cpu_startup_entry+0x1f/0x21
[  268.562357]  start_secondary+0x152/0x170
[  268.562362]  secondary_startup_64+0xa4/0xb0
[  268.562367] ---[ end trace 8110ae1151933f03 ]---
[  269.932759] rtw_8822ce 0000:02:00.0: timed out to flush queue 1
[  270.042419] rtw_8822ce 0000:02:00.0: timed out to flush queue 2
[  270.153392] rtw_8822ce 0000:02:00.0: timed out to flush queue 1
[  270.263317] rtw_8822ce 0000:02:00.0: timed out to flush queue 2
```

mingcheng, 2022-09-01:
try to add following into your kernel params 

`rtw88_pci.disable_aspm=true rtw88_core.lps_deep_mode=0`

see https://bbs.archlinux.org/viewtopic.php?id=266359

drujd, 2022-11-23:
I have exactly the same problems on Steam Deck (with RTL8822CE soldered on), which uses the rtw88 driver. It happens when connected to QCA QCA9880 AP (in 5GHz vht80 mode). I tried the "disable wi-fi power management" option in Developer Options, seems to slightly decrease the likelihood of the "timed out to flush queue x" to happen, but it still happens.

s-b-repo, 2024-05-13:
rtw_8821cu 5-1:1.0: timed out flash queue 2 same issue with kali