sebanc/brunch #857: intel celeron N3350 not booting "Chrome os was not found on any device, falling back to shell"

diegosky1405, 2021-02-08:
Tried installing several times following the guide here, redownloaded everything in case somethings is broken and also i posted this same issue on discord too , i just get the message "Chrome os was not found on any device, falling back to shell..." . 

### This is my grub after installing brunch

```

set timeout=2

menuentry "ChromeOS" {
    regexp --set disk "(^.+)(,gpt)" $root
    linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug \
          console= vt.global_cursor_default=0 brunch_bootsplash=default
    initrd ($disk,7)/lib/firmware/amd-ucode.img ($disk,7)/lib/firmware/intel-ucode.img ($disk,7)/initramfs.img
}
menuentry "ChromeOS (debug mode)" {
    regexp --set disk "(^.+)(,gpt)" $root
    linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug
    initrd ($disk,7)/lib/firmware/amd-ucode.img ($disk,7)/lib/firmware/intel-ucode.img ($disk,7)/initramfs.img
}

```

### This is the full console log when i install 

```

mint@mint:/cdrom/Chrome OS$ sudo bash chromeos-install.sh -src rammus_recovery.bin -dst /dev/mmcblk1
All data on device /dev/mmcblk1 will be lost, are you sure ? (type yes to continue) yes
umount: /dev/mmcblk1: not mounted.
umount: /dev/mmcblk1boot0: not mounted.
umount: /dev/mmcblk1boot1: not mounted.
umount: /dev/mmcblk1p1: not mounted.
umount: /dev/mmcblk1p10: not mounted.
umount: /dev/mmcblk1p11: not mounted.
umount: /dev/mmcblk1p12: not mounted.
umount: /dev/mmcblk1p2: not mounted.
umount: /dev/mmcblk1p3: not mounted.
umount: /dev/mmcblk1p4: not mounted.
umount: /dev/mmcblk1p5: not mounted.
umount: /dev/mmcblk1p6: not mounted.
umount: /dev/mmcblk1p9: not mounted.
umount: /dev/mmcblk1rpmb: not mounted.
26E537E6-0D26-0646-A23C-28B74A9579F5
       start        size    part  contents
           0           1          PMBR (Boot GUID: 26E537E6-0D26-0646-A23C-28B74A9579F5)
           1           1          Pri GPT header
           2          32          Pri GPT table
    19255304    41816008       1  Label: "STATE"
                                  Type: Linux data
                                  UUID: 78868E72-5531-2F49-9695-BC7392DEFA39
     2117638       65536       2  Label: "KERN-A"
                                  Type: ChromeOS kernel
                                  UUID: D85F243E-0F29-564E-8E3B-07D5201B4DA1
                                  Attr: priority=15 tries=15 successful=0 
    10866696     8388608       3  Label: "ROOT-A"
                                  Type: ChromeOS rootfs
                                  UUID: 6211B781-69BF-7D44-9293-C37AB709B5FC
     2183174       65536       4  Label: "KERN-B"
                                  Type: ChromeOS kernel
                                  UUID: E6E2BF01-A9E7-2C44-AA0A-B2809C822CC0
                                  Attr: priority=0 tries=15 successful=0 
     2478088     8388608       5  Label: "ROOT-B"
                                  Type: ChromeOS rootfs
                                  UUID: A158B919-E630-C244-9D50-0ECF19B6DF79
       16448           1       6  Label: "KERN-C"
                                  Type: ChromeOS kernel
                                  UUID: AA966CF3-7A55-2D47-A7A8-92726677D1EF
                                  Attr: priority=0 tries=15 successful=0 
       16456     2097152       7  Label: "ROOT-C"
                                  Type: ChromeOS rootfs
                                  UUID: 25D190E5-7121-044D-A383-684AB94E7B33
     2248712       32768       8  Label: "OEM"
                                  Type: Linux data
                                  UUID: 480C5B13-94DE-754B-AF93-732CED664106
     2113608           1       9  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: CBBD8EC7-5C11-FD45-83A4-E3B928E7470B
     2113609           1      10  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: FD286AA2-2D8D-D447-B9D6-F894274D9BF4
          64       16384      11  Label: "RWFW"
                                  Type: ChromeOS firmware
                                  UUID: FB4FD501-26F4-3344-A2DF-770792D4F2E4
     2412552       65536      12  Label: "EFI-SYSTEM"
                                  Type: EFI System Partition
                                  UUID: 26E537E6-0D26-0646-A23C-28B74A9579F5
    61071327          32          Sec GPT table
    61071359           1          Sec GPT header
Writing partition 1
8.95MiB 0:00:00 [20.5MiB/s] [=======================================================================>] 100%            
Writing partition 2
32.0MiB 0:00:01 [19.8MiB/s] [=======================================================================>] 100%            
Writing partition 3
2.34GiB 0:02:00 [19.9MiB/s] [=======================================================================>] 100%            
Writing partition 4
32.0MiB 0:00:01 [19.8MiB/s] [=======================================================================>] 100%            
Writing partition 5
2.34GiB 0:01:58 [20.2MiB/s] [=======================================================================>] 100%            
Writing partition 6
 512 B 0:00:00 [2.43MiB/s] [========================================================================>] 100%            
Writing partition 7
1.00GiB 0:00:48 [21.0MiB/s] [=======================================================================>] 100%            
Writing partition 8
16.0MiB 0:00:00 [20.8MiB/s] [=======================================================================>] 100%            
Writing partition 9
 512 B 0:00:00 [2.26MiB/s] [========================================================================>] 100%            
Writing partition 10
 512 B 0:00:00 [ 217KiB/s] [========================================================================>] 100%            
Writing partition 11
8.00MiB 0:00:00 [21.3MiB/s] [=======================================================================>] 100%            
Writing partition 12
32.0MiB 0:00:01 [20.7MiB/s] [=======================================================================>] 100%            
ChromeOS installed.

```

### And this are my specs :

Processor: Intel celeron N3350 1.10GHZ x2
Memory: 2GB
Disk: 32GB Emmc

diegosky1405, 2021-02-08:
ill be happy to provide anything else that may be required :)

sebanc, 2021-02-08:
I saw the different logs you posted on the Discord group and it seems to be some kind of emmc issue.

Could you boot brunch (with rammus recovery image) to the point where it says "chromeos not found on any device" and provide the output of the below commands:
```
blkid -s PARTLABEL -o value /dev/mmcblk1p1
blkid -s PARTLABEL -o value /dev/mmcblk1p3
```

diegosky1405, 2021-02-08:
```

blkid -s PARTLABEL -o value /dev/mmcblk1p1 ->Does nothing
blkid -s PARTLABEL -o value /dev/mmcblk1p3 -> "ROOT-A"
if [ -b /dev/mmcblk1p1 ]; then echo "ok"; fi -> "ok"
if [ -b /dev/mmcblk1p3 ]; then echo "ok"; fi -> "ok"

```

sebanc, 2021-02-08:
What does "mkfs.ext4 -F -b 4096 -L "H-STATE" /dev/mmcblk1p1" returns in linux ?

sebanc, 2021-02-08:
Sorry forget the previous answer, what does "sudo cgpt show /dev/mmcblk1" returns ?

sebanc, 2021-02-08:
and what linux distro are you using ?

diegosky1405, 2021-02-08:
> Sorry forget the previous answer, what does "sudo cgpt show /dev/mmcblk1" returns ?

sh: sudo: not found

diegosky1405, 2021-02-08:
> and what linux distro are you using ?

to install brunch you mean ? i use linux mint from usb bootable

diegosky1405, 2021-02-10:
Updating this for anyone having the same issue, after correctly installing brunch you need to run this line on cmd  "sudo mkfs.ext4 -F -b 4096 -L "H-STATE" /dev/mmcblk1p1" , after that just reboot and system boots now. Of course full credits to @sebanc for the quick fix :)