sebanc/brunch #859: Randomly reboots

geo1925, 2021-02-09:
Installed from linus mint on old intel 4th generation processor.
menuentry is `****************************
menuentry "ChromeOS (boot from disk image)" {
	rmmod tpm
	img_part=/dev/sda5
	img_path=/TestingOS.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path \
		console= vt.global_cursor_default=0 brunch_bootsplash=default
	initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}`
Chrome OS reboots after `ChromeOS rootfs patches are being applied`
Waiting for help

geo1925, 2021-02-09:
I have inbuilt graphics.

geo1925, 2021-02-09:
Please HELP

sebanc, 2021-02-09:
Could you remove "console= vt.global_cursor_default=0 brunch_bootsplash=default" from the grub config and the "\" at the end of the previous line and post a short video of the boot process ?

ameyjoshi3068, 2021-02-10:
> Could you remove "console= vt.global_cursor_default=0 brunch_bootsplash=default" from the grub config and the `\` at the end of the previous line and post a short video of the boot process ?

Sure.
Thanks for replying

khanhtranngoccva, 2021-02-10:
Can you try to install it on a SSD?