sebanc/brunch #860: Issues regarding Acer Nitro 5

neptalactone, 2021-02-09:
Hi, I have tried install chromeOS using brunch method on this laptop, but i have encountered problem. I write the chronologies of what i have been tried and what my laptop specs is.

First my laptop specs are using AMD Ryzen 5 3550H as CPU, from what i read in this repo readme, Ryzen 3XXX series are supported with zork recovery. so i went with it, but i also tried to use the brunch-tool-kit to check my CPU compatibility with brunch and chromeOS, and it's says it might be supported and it recommends me grunt recovery, so i download both recovery. 

I have tried installing brunch with multiboot method on my linux machine. creating the partition and build the chromeOS image to boot from following the guide on multiboot section with linux in the readme. First, i was using zork recovery. I able to make brunch boot and waiting for it to complete rebuilding the rootfs. after it's done i've been greeted by chrome logo. I thought i made it. but the chrome logo won't disappear even when i have waited for 1 hour. it's stuck there. i tried pressing ctrl+alt+f2 and i'm able to switch to CLI (developer mode?) but i can't do anything in there. so i reboot, change the grub config to tell brunch to change the default kernel to kernel 5.10 and using android_init_fix options. But it also didn't resolve the problem. 

Having that issue, then i decided to change the recovery image that i use from zork to grunt. because the toolkit recommend that image. Using the same process, i also able to boot to brunch, then wait for brunch complete it's rebuilding rootfs process. after it's finished, it's reboot my laptop. i tried to boot to brunch again, but it rebooting my system again. i back to my linux machine, change the grub config and kernel parameter to include kernel 5.10 and android_init_fix options. then boot again. But the results still the same, i'm still having bootloop. 

Having those two issue, i tried other brunch release, r88, and both r87. but the issue still the same. i'm stuck on chrome logo in zork recovery. and i'm having bootloop with grunt recovery.

Here's the TL;DR version.
- I tried installing ChroeOS multiboot with my linux
- I have AMD Ryzen 5 3550H CPU, so i went with zork, but toolkit recommend me grunt, so i download both
- I use r87 brunch release following the guide (brunch release version = recovery version)
- I build the boot image using brunch r87 and zork recovery version 87, able to boot to brunch, brunch rebuilding process complete, but stuck on chrome logo
- I change the grub config to make brunch use kernel 5.10 rather than default and add android_init_fix options, update grub, wait brunch rebuilding again, still stuck on chrome logo
- Then i change the recovery image with grunt recovery verion 87, able to boot to brunch, brunch rebuilding process complete, but having bootloop problem
- I also try to change grub config too as the same as before, update grub, wait brunch rebuilding again, but still having bootloop
- Then i tried other brunch release, r88 and another r87 release. Still having the same problem

I just wanna share this experience in case someone with the same hardware want to install ChromeOS with brunch.
I appreciate if you guys know how to fix the issue i have.

sebanc, 2021-02-09:
Is your computer a desktop or laptop ? Do you have a nvidia gpu or just amd ?

neptalactone, 2021-02-10:
> Is your computer a desktop or laptop ? Do you have a nvidia gpu or just amd ?

It is a laptop. Here's the complete specification:
CPU: AMD Ryzen 5 3550H
GPU: AMD Radeon Vega 8 mobile and AMD Radeon RX 550
Realtek RTL8111/8168/8411 Gigabit Ethernet
Qualcomm Atheros QCA6174 802.11ac wireless adapter

sebanc, 2021-02-10:
The issue is that chromeos is booted but on the gpu which is not connected directly to the display.

Try disabling one in your bios if you can or alternatively you can try adding "fbcon=map:1" to the kernel command line.

neptalactone, 2021-02-10:
> The issue is that chromeos is booted but on the gpu which is not connected directly to the display.
> 
> Try disabling one gpu in your bios if you can or alternatively you can try adding "fbcon=map:1" to the kernel command line.

Oh, i never thought about it, This laptop didn't have such setting on its BIOS, I'll try the kernel command line. I'll update if it's work or not. Thank you

neptalactone, 2021-02-10:
> The issue is that chromeos is booted but on the gpu which is not connected directly to the display.
> 
> Try disabling one gpu in your bios if you can or alternatively you can try adding "fbcon=map:1" to the kernel command line.

Hi, i already try it with zork recovery image. I still got stuck at chrome logo even with adding your recommended kernel command line and changing the default kernel to kernel 5.10. i try press ctrl+alt+f2, i get CLI. i login with default login, try `uname -a` and it gives my exact CPU. which i think probably my CPU is supported. the result of uname command goes like this:

`Linux localhost 5.10.0-brunch-sebanc #1 SMP PREEMPT Sun Dec 09:20:33 CET 2020 x86_64 AMD Ryzen 5 3550H with Radeon Vega mobile `

neptalactone, 2021-02-10:
I have access to the journalctl with root on the chrome with CLI (developer mode?). i found several error that might be the reason why i got stuck on the chrome logo. [Here ](https://pastebin.com/Njccy3S6) the snippet of the journal log that the error occured. If you want the full log, i save it on my drive [here](https://drive.google.com/file/d/1pDXBbm2OrvdwFbe_9i_xZ7FbHTPTtQOw/view?usp=sharing).

One thing that interesting to me in that log file is, seems like somehow it use the dGPU driver (POLARIS11, AMD Radeon RX 550), rather than using the iGPU driver (RAVEN, AMD Radeon Vega 8) that comes with CPU (AMD Ryzen 5 3550H). Like this image:

[image]

RAVEN is the iGPU driver and POLARIS11 is the dGPU driver. And chromeOS somehow using POLARIS11 driver as default.

sebanc, 2021-02-10:
Indeed, the issue seems to be that the wrong gpu is used. The only kernel command line argument I know about which seems to help switching gpu is "fbcon=map:1" (you can try other values). Aside from that I have no idea how to select the gpu if not by disabling one in the bios.

neptalactone, 2021-02-11:
> Indeed, the issue seems to be that the wrong gpu is used. The only kernel command line argument I know about which seems to help switching gpu is "fbcon=map:1" (you can try other values). Aside from that I have no idea how to select the gpu if not by disabling one in the bios.

Yeah, i try other kernel parameters from amdgpu documentation but none working. I try the command line from yours with 0 but the results still same. Well i guess ChromeOS cannot run well in this machine.

neptalactone, 2021-02-11:
Should i leave the issue open or not?

neptalactone, 2022-03-18:
> How you changed and SAVED the grub config to make brunch use kernel 5.10?

I forget how i do that. but that's a while back. @sebanc @xtibrox 

Right now, i tried too boot brunch again. Before i tried brunch again, i was trying to install FydeOS with their latest AMD build, i managed to get by their boot logo, but then i got greeted by black screen and a mouse cursor. I can move the cursor and sometimes the mouse cursor change icon, indicating there's something that i can click on there. 

After i failed to boot FydeOS, I'm trying brunch again. This time there is an advancement than before. Before i get stuck on ChromeOS logo. But This time i get to pass that, but after the logo i have encounter the same issue like FydeOS with brunch. After ChromeOS logo, it will give me a black screen but with a mouse cursor, that i can move and see some icon cursor changes. Like this image:
https://imgur.com/88CChwR and https://i.imgur.com/t2TkC5L.jpg

Curious, i open tty and login as root. and try to see the output of the last 200 line of dmesg. This is the dmesg output:
https://i.imgur.com/LLDB4IS.jpg
when i saw that, I remembered the kernel parameter that @sebanc told me to add, so I added the "fbcon:map=1" kernel parameter through the "ChromeOS Settings" option on the new grub loader. i also try to change the kernel to 5.10 and 5.15 both with the same kernel parameter.

After I'm done, the issue still persists, it still gives me a black screen with a mouse cursor that i can control. I go back to tty and see the dmesg output. this time, the dmesg error message about the framebuffer didn't show up. Having stuck in this problem both on brunch ChromeOS and FydeOS. I'm thinking of giving ChromeOS Flex a try. 

Without having any expectation that it will boot. To my surprise, ChromeOS Flex boots and works fine. All of it works fine, the Wifi, Bluetooth, Display (resolution), Speaker, Mic, Touchpad. One problem tho, is that it won't wake up from suspend/sleep. but again i tried ChromeOS Flex from USB (live environment) rather than installing it on my drive. I check which GPU ChromeOS Flex uses from the chrome://gpu flag. and here is the output:
https://i.imgur.com/qpNqaWj.jpg and https://i.imgur.com/f7zlGkB.jpg

From the output, it seems like it correctly use my integrated gpu (vega 8) rather than the dedicated ones (rx 550). Now the catch is, i noticed some patch on brunch when i activate brunch with verbose output, it applied a patch with a name "dgpu.sh". what patch is that? perhaps it is because of that patch that somehow brunch use different gpu for the display rather than the correct one.

Any insight of this? @sebanc @xtibrox. I'm hoping that from my findings it could expand brunch or even FydeOS compatibility hardware.

EthraZa, 2022-07-21:
Today I spent the day and ended up with the same results as you @neptalactone with damn Nitro 5. FydeOS give black screen with cursor. Chrome OS zork/ezkinil + brunch give bootloop. after showing Chrome OS logo. Flex works on the pen drive, but not sure if I can get it installed to SSD  dual booting as well. Guess I gonna try it now.

neptalactone, 2022-07-27:
Just a little update. I tried again to install brunch ChormeOS with brunch version 101 and ChromeOS version 101. It is somehow fixed now and i am able to boot into ChromeOS and get the desktop. Everything works perfectly now. I can use Linux Development Environment on ChromeOS, using android app, and even i install Broiche (which is crouton for brunch ChromeOS). 

The only caveat here is, for my device especially, i have to choose internal_microphone_fix_2 to make my internal microphone functional. and sometime android app won't load at all so i have to insert `chrome://restart` to use android app again.

I suggest you @EthraZa try brunch again but with version 101. i tried with version 102 and 103, somehow it have many issue again. like blackscreen and bootlooping. As for FydeOS, i think this fix applied on ChromeOS version 101. and FydeOS right now is still based on version 96.

EthraZa, 2022-07-27:
Thank you for the followups. 
In the Flex side, I have managed to install it by disabling one HD in the BIOS and reenabling it after Flex installation is complete. I'm using F12 boot menu to choose where to boot, but I saw I can use rEFInd for easy boot menu, I'm just too lazy now. I have installed Android-x86 under Linux QEMU/Virt Manager, it works but is extremely slow, even if it looks like is using KVM.