sebanc/brunch #866: Stuck at Chrome OS logo - MacBook Pro (15-inch, 2018)

tpnonthealps, 2021-02-10:
Not sure if it is related to the Radeon Pro 560X graphics card (maybe related to #554 and #707. However, I am not using an Nvidia graphics card).

On bootcamp Windows, the graphics card requires a special AMD driver made for bootcamp devices (https://www.amd.com/zh-hant/support/kb/release-notes/apple-boot-camp), and the generic AMD driver cannot be used. However, the `amdgpu` kernel on Linux seems to be able to make the graphics card work properly.

What should I do?

sebanc, 2021-02-10:
Have you tried "zork" recovery image ?

tpnonthealps, 2021-02-10:
> Have you tried "zork" recovery image ?

But the CPU of my MacBook Pro is `Intel i7-8850H (12) @ 2.60GHz`, should I give it a try?

sebanc, 2021-02-10:
I thought you did not have an igpu but actually according to your cpu specs it should have an igpu so "rammus" recovery image should work.

Otherwise zork might give an interesting result.

tpnonthealps, 2021-02-10:
Additional information here: My CPU is `Intel i7-8850H (12) @ 2.60GHz`, and my GPU is `Intel UHD Graphics 630` and `Radeon Pro 560X`. Should I use the solution mentioned in #554 to force the Intel GPU to be loaded and blacklist the AMD GPU from booting? Or should I give a try to the `zork` recovery image?

sebanc, 2021-02-10:
Yes, that's a very good idea. Try adding "module_blacklist=amdgpu" on the kernel command line

sebanc, 2021-02-10:
zork recovery image is more an experiment, someone with more or less the same hardware was able to boot zork which was quite surprising but "module_blacklist=amdgpu" is a much better idea.

closed 2026-03-17 by github-actions[bot]