sebanc/brunch #869: First boot os so fast, second is slow slow

NR-No-Rage, 2021-02-11:
Config:
Pc: Acer aspire 3
Cpu: rysen 3
Gpu: readon Vegas 3
Ram: 8G
Hdd: mvne 500G
OS: dualboot With grub2win: ChromeOS (Zork 87 With brunch r88) And windows 10 LTSC

Hello everybody, I don't know anything about Linux but really nothing.

So I put Linux Mint on my USB stick, I start on Linux Mint live To install Chrome OS on my mvne

Once I have followed the whole procedure to install Chrome OS

I go to Chrome OS it's super smooth and super fast when I launch Google Play applications, very quickly it responds to the click, with no lag a lot of fps, But when I restart my computer to go to Chrome OS it becomes super slow the applications take a long time to launch and a lot of lag on chrome os and apk

If you need more explanation ask me

sebanc, 2021-02-12:
The only difference I can think of is that the android media scanner is not active on first boot. If you have an sdcard or usb flashdrive with a lot of files in it, try waiting for the android media scanner to finish scanning your device.

For that reason the "mount_internal_drives" option is not recommended (not sure if you use it or not)

NR-No-Rage, 2021-02-15:
No i dont use (mount_internal_drives) I installed it on a 500 GB m2 .i go  will test on a 30 GB partition.