sebanc/brunch #875: Looking for Chrome OS r88 recovery images for testing "Brunch r88 stable 20210131"

johnnylau34, 2021-02-12:
Hi all,

If I want to test drive "Brunch r88 stable 20210131" on my Samsung Flex 2 5G with OLED display, where can I locate a r88 recovery image? I can only locate r87 at https://cros.tech/device/rammus

Thanks!

Best Regards,
Johnny

sebanc, 2021-02-12:
use a chromeos r87 recovery image, brunch r88 is backwards compatible.

If you have an oled display, you will want to use kernel 5.10.

Tash254-ke, 2021-02-13:
I am currently on chrome os r87 with brunch r87, if I put brunch r88, when chrome os r88 is available can I just use sudo chromeos-update -r < path to the ChromeOS recovery image > to update without including the brunch part?
Sorry if this is a stupid question.

b4iterdev, 2021-02-14:
maybe you'd rather update brunch to lastest version first, then enable native update. Because r88 recovery image is only available when r89 is released, if you want to wait, update both brunch and os to r88 with ``` sudo chromeos-update -r < path to the ChromeOS recovery image > -f < path to the Brunch release archive > ```

b4iterdev, 2021-02-14:
If your problem is solved, consider close this issue

Tash254-ke, 2021-02-14:
> maybe you'd rather update brunch to lastest version first, then enable native update. Because r88 recovery image is only available when r89 is released, if you want to wait, update both brunch and [...]

When I enable native update, can I disable it later? Because chrome os auto updates, wouldn't there be an issue when I am on brunch r88 and chrome os has auto updated to r89?

b4iterdev, 2021-02-16:
> When I enable native update, can I disable it later? Because chrome os auto updates, wouldn't there be an issue when I am on brunch r88 and chrome os has auto updated to r89?

of course you can disable it

Tash254-ke, 2021-02-16:
> of course you can disable it

How?

b4iterdev, 2021-02-17:
> How?

just remove enable_updates in option, rootfs will be rebuilt automatically.