sebanc/brunch #876: i have a problem installing dual chrome os i get this error.

djgerman10, 2021-02-12:
i have a problem installing dual chrome os i get this error.

[image]

[image]

[image]

NR-No-Rage, 2021-02-12:
U have installed Chromebrew?

kushmittal2009, 2021-02-13:
> i have a problem installing dual chrome os i get this error.
> 
> [image]
> 
> [image]
> 
> [image]

@djgerman10 Make sure you use the chrome-install.sh file that sebanc provided then just ignore this message. Boot into windows 10. Then open explorer and go to where you made the chrome os image. There you will see another file named something like grub. Open it. Copy all the contents inside the stars. Then install grub 2 win. Add a new configuration custom code and paste that code there. On next boot select chrome os and done!

closed 2021-02-13 by djgerman10