sebanc/brunch #877: support for ASIX usb ethernet adapters

bobafetthotmail, 2021-02-12:
chipsets AX8817X from ASIX are the "better" kind of USB ethernet adapters (Gigabit or 100Mbit depending on what the last X in the name is) that also work well on Linux and on the stock ChromeOS firmware.

They are supported by upstream Linux drivers **USB_NET_AX8817X** and **USB_NET_AX88179_178A**

I don't see these symbols in your source, and my dongles don't work. 

Can you add these to the kernel in brunch? 

It would be very nice to be able to use these dongles, my Dell Tablet (that otherwise works great with brunch/ChromeOS) has no ethernet port but it has a USB 3.0 port I could use these adapters in

sebanc, 2021-02-14:
Those drivers are enabled in the brunch config so I am not sure why it does not work...

Maybe a firmware issue, what's the output of "dmesg | grep -i firmware" in crosh shell (with the adapter plugged) ?

bobafetthotmail, 2021-02-14:
I think ASIX dongles do not need a firmware load.
I don't see anything in dmesg when I plug or unplug it.
there is nothing useful in "dmesg | grep -i firmware" either, I only get lines about TPM bugs on my platform, Spectre vulnerability and that the firmware for the intel wifi/bluetooth card was loaded correctly.

 I'm using brunch 87 at the moment.

This is the dmesg from my PC running OpenSUSE (where the dongle work correctly) as an example of what I expected in dmesg when connecting it.
```
[238631.699780] usb 4-2: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[238631.726127] usb 4-2: New USB device found, idVendor=0b95, idProduct=1790, bcdDevice= 1.00
[238631.726131] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[238631.726134] usb 4-2: Product: AX88179
[238631.726136] usb 4-2: Manufacturer: ASIX Elec. Corp.
[238631.726137] usb 4-2: SerialNumber: 00000000000179
[238632.781084] ax88179_178a 4-2:1.0 eth0: register 'ax88179_178a' at usb-0000:0b:00.3-2, ASIX AX88179 USB 3.0 Gigabit Ethernet, 00:0e:c6:70:2a:22
[238632.781166] usbcore: registered new interface driver ax88179_178a
[238632.803064] ax88179_178a 4-2:1.0 enp11s0f3u2: renamed from eth0
```

sebanc, 2021-02-15:
Maybe try updating brunch, all I can say is that the driver is there and your usb product / vendor ID are correctly present in it.

I guess plugging in the adapter should anyway display some usb lines, is it on the same machine that you are testing the adapter on ?

bobafetthotmail, 2021-02-15:
actually a different issue, opened new issue  about it here https://github.com/sebanc/brunch/issues/889

closed 2021-02-15 by bobafetthotmail