sebanc/brunch #878: broadcom_wl does not work since February 2021 update any more

weberk, 2021-02-12:
```
$ cat /etc/brunch_version
Brunch r87 20201227
$ Chormeos rammus version:
Version 88.0.4324.109 (Offizieller Build) (64-Bit)

```
dmesg:
```
[   13.233842] wl: module license 'MIXED/Proprietary' taints kernel.
[   13.241523] pci 0000:03:00.0: attach allowed to drvr wl [trusted device]
[   13.306025] ERROR @wl_alloc_wdev : 
[   13.306026] Couldn not register wiphy device (-22)
[   13.306042] ERROR @wl_cfg80211_detach : 
[   13.306043] NULL ndev->ieee80211ptr, unable to deref wl
[   16.515889] init: broadcom_wl main process (1880) terminated with status 1

```
lspci shows no driver in use:

```
03:00.0 Network controller: Broadcom Corporation BCM4360 802.11ac Wireless Network Adapter (rev 03)
        Subsystem: Apple Inc. BCM4360 802.11ac Wireless Network Adapter
        Flags: bus master, fast devsel, latency 0, IRQ 18
        Memory at b0600000 (64-bit, non-prefetchable) [size=32K]
        Memory at b0400000 (64-bit, non-prefetchable) [size=2M]
        Capabilities: [48] Power Management version 3
        Capabilities: [58] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [68] Vendor Specific Information: Len=44 <?>
        Capabilities: [ac] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [13c] Device Serial Number 84-38-00-ff-ff-00-00-01
        Capabilities: [150] Power Budgeting <?>
        Capabilities: [160] Virtual Channel
        Capabilities: [1b0] Latency Tolerance Reporting
        Capabilities: [220] #15
        Kernel modules: bcma, broadcom_wl
```

my kernel option:
`options=broadcom_wl,enable_updates,acpi_power_button
`
Any idea howto recover?

weberk, 2021-02-13:
Back to working after ChromeOS update to Version 88.0.4324.153 (Offizieller Build) (64-Bit)

weberk, 2021-02-13:
WIFI driver always breaks after I have executed
depmod -A or  depmod -a

Brax84, 2021-02-16:
@weberk 

Try this [my fix](https://github.com/sebanc/brunch/issues/883#issuecomment-778831479)

weberk, 2021-02-17:
My situation seems different than yours. wlan0 works at full speed.
When I try to deploy a self compiled kernel module wlan0 stops working at all. 
In order to allow modprobe to load the new kernel module it needs to call depmod first. The latter than breaks broadcom_wl

closed 2026-03-15 by github-actions[bot]