sebanc/brunch #880: Surface Pro 7 touchscreen not working.

kiluae, 2021-02-13:
On newest stable brunch, I've added options=ipts after "cros_debug".

Everything else seems to work, only touch screen is completely nonfunctional.

thorstenludwigs, 2021-02-13:
Hi 
here is my working config Brunch r88 stable 20210131 & rammus 87.0.4280.152 (Surface Pro 7 i5 8GB RAM)

menuentry "ChromeOS" {
        regexp --set disk "(^.+)(,gpt)" $root
        linux ($disk,7)/kernel boot=local noresume noswap options=ipts,advanced_als loglevel=7 disablevmx=off cros_secure cros_debug \
                console= vt.global_cursor_default=0 brunch_bootsplash=default
        initrd ($disk,7)/lib/firmware/amd-ucode.img ($disk,7)/lib/firmware/intel-ucode.img ($disk,7)/initramfs.img
}

rexsububi, 2021-02-13:
I have the same issue. Just changed my config to the same as @thorstenludwigs and can confirm that my touch screen is still nonfunctional.

closed 2021-02-14 by kiluae