sebanc/brunch #881: how to change kernel?

djgerman10, 2021-02-13:
I have **intel bay trail**, I want to put the kernel 4.19 because it doesn't work at all well.

[image]

_I am lost, someone who can help me?_

b4iterdev, 2021-02-14:
You are dual-booting, so if you are using grub2win, boot into windows, open grub2win app, manage boot order, click chrome os, then click in boot config file, notepad will show up then you can change the kernel ( https://github.com/sebanc/brunch#changing-kernel-version )

b4iterdev, 2021-02-14:
If your problem has solved, please consider close this issue.

closed 2021-02-14 by djgerman10