sebanc/brunch #883: Macbook pro 15" 2011 dim brightness screen

Brax84, 2021-02-13:
Hi,

first of all thanks for your great work!
My macbookpro work very well but the only problem, is wich not possible to dim the brightness of monitor.
On linux mint works the dim. It's possibile to have the module kernel in chromeos for to work this option?

In attach the list of :
lsmod of chromeos 
lsmod of Linux Mint

Thanks
[lsmod.zip](https://github.com/sebanc/brunch/files/5976259/lsmod.zip)

sebanc, 2021-02-14:
The kernel module is included (else you could not change it) but the brightness is not at a standard path so chromeos does not see it.

What I suggest is to make a bootscript to set the brightness on boot:
```
sudo mkdir -p /var/brunch/bootscripts
echo "echo 10000 > /sys/class/backlight/gmux_backlight/brightness" | sudo tee /var/brunch/bootscripts/brightness.sh
sudo chmod 0755 /var/brunch/bootscripts/brightness.sh
```

Brax84, 2021-02-14:
> but the brightness is not at a standard path so chromeos does not see it.

Ah OK! Thank you very much !

Two more questions ...

Wi-Fi is slow at most it goes to 15 Mbps, with mac or Linux the maximum speed is 100 Mbps correct.
Adapter is BCM94360CS2

Sudo iw list show 

>         Band 1:
>                 Bitrates (non-HT):
>                         * 1.0 Mbps
>                         * 2.0 Mbps (short preamble supported)
>                         * 5.5 Mbps (short preamble supported)
>                         * 11.0 Mbps (short preamble supported)
>                         * 6.0 Mbps
>                         * 9.0 Mbps
>                         * 12.0 Mbps
>                         * 18.0 Mbps
>                         * 24.0 Mbps
>                         * 36.0 Mbps
>                         * 48.0 Mbps
>                         * 54.0 Mbps
>                 Frequencies:
>                         * 2412 MHz [1] (20.0 dBm)
>                         * 2417 MHz [2] (20.0 dBm)
>                         * 2422 MHz [3] (20.0 dBm)
>                         * 2427 MHz [4] (20.0 dBm)
>                         * 2432 MHz [5] (20.0 dBm)
>                         * 2437 MHz [6] (20.0 dBm)
>                         * 2442 MHz [7] (20.0 dBm)
>                         * 2447 MHz [8] (20.0 dBm)
>                         * 2452 MHz [9] (20.0 dBm)
>                         * 2457 MHz [10] (20.0 dBm)
>                         * 2462 MHz [11] (20.0 dBm)
>                         * 2467 MHz [12] (20.0 dBm)
>                         * 2472 MHz [13] (20.0 dBm)
>                         * 2484 MHz [14] (disabled)
> 
> 

Last question...
Is it possible to set the fan speed from the shell?

sebanc, 2021-02-14:
The command to set the fan speed would be the same than on linux, you might want to browse the linux forums.

WRT wifi, could you post a dmesg ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and post dmesg.txt from your Downloads folder)

sebanc, 2021-02-14:
You can also try to run:
`sudo iw dev wlan0 set power_save off`

Brax84, 2021-02-14:
@sebanc 

> The command to set the fan speed would be the same than on linux, you might want to browse the linux forums.

Ok thanks!

`sudo iw dev wlan0 set power_save off
`

Same speed result 

Attach dmesg.txt
[dmesg.txt](https://github.com/sebanc/brunch/files/5977647/dmesg.txt)

sebanc, 2021-02-14:
Sorry the correct command was "sudo iw wlan0 set power_save off"

Brax84, 2021-02-14:
> Sorry the correct command was "sudo iw wlan0 set power_save off"

Many thanks, but same problem stuck at max 15Mbps
Same message in dmesg `Process '/usr/sbin/iw dev wlan0 set power_save on'
[dmesg.txt](https://github.com/sebanc/brunch/files/5978312/dmesg.txt)
 failed with exit code 161
`

Brax84, 2021-02-14:
@sebanc 
I've solve slow wifi by this way:

sudo edit-grub-config
module_blacklist=ssb,b43,bcma

then reboot

sudo edit-grub-config
options=broadcom_wl

then reboot

Tada...wifi work at max speed of my ISP  and the 5Ghz is finally active!

Many many thanks for your fast support and for you wonderful work !

Best regards by Italy

closed 2021-02-16 by Brax84