sebanc/brunch #885: Scrolling stutter / lag on Brunch r88 stable 20210131

mtimms1980, 2021-02-14:
Firstly, amazing work! I am happily triple booting Win10, macOS and ChromeOS and of all 3 ChromeOS feels the best on my MateBook X Pro 2019!

I noticed recently that scrolling in all areas took a very apparent hit in terms of smoothness, so I downgraded to Brunch r86 k4.19 stable 20201014 and things are nice and smooth again. I am running ChromeOS 88 so I realise I may run into issues with that, but for now at least this is the better combination for me. 

It may not be the kernel that has caused it of course, anyone else seeing this?

khanhtranngoccva, 2021-02-15:
Did you mean the Android apps? Try updating to Beta and powerwashing once, this works perfectly for me

mtimms1980, 2021-02-15:
No I meant throughout the whole OS, most noticeable in web browser and settings app.

closed 2026-03-15 by github-actions[bot]