sebanc/brunch #889: USB 3.0 not working on Dell Venue 11 7130

bobafetthotmail, 2021-02-15:
the SoC is a Intel Core i3-4020Y, and this tablet has a USB 3.0 port on the side.

The USB 3.0 port works only in 2.0 mode, 3.0 devices are not even detected.

I tried plugging in some USB 2.0 or lower devices like a wireless keyboard/mouse combo from Logitech, a USB flash drive and a card reader, they all show up in dmesg and work correctly.

I tried plugging in a USB 3.0 hard drive, a USB 3.0 ethernet dongle and a USB 3.0 card reader, none of these devices are detected, although I see they are powered.

If I swap the SSD and boot OpenSUSE linux, the USB 3.0 port is working fine.

I also see that every time I connect a USB 2.0 device I get an error in the dmesg

```
udevd[xxxx] failed to execute 'usr/sbin/ectool' 'usr/sbin/ectool usbchargemode 0 2 0': Exec format error
udevd[xxxx] process 'usr/sbin/ectool usbchargemode 0 2 0' failed with exit code 2
```

sebanc, 2021-02-20:
Could you attach the output of "lsmod" and "dmesg" from linux with the adapter plugged ?

bobafetthotmail, 2021-02-21:
attached files
[lsmod.txt](https://github.com/sebanc/brunch/files/6018067/lsmod.txt)
[dmesg.txt](https://github.com/sebanc/brunch/files/6018068/dmesg.txt)

bobafetthotmail, 2021-02-21:
seems like the whole USB 3.0 controller driver (**xhci_hcd**) is missing, or at least not compiled as a module so it does not show up in lsmod.

On OpenSUSE (where the USB 3.0 port works properly) this driver is a module and is loaded, I have these lines about it in lsmod 

```
xhci_pci               20480  0
xhci_pci_renesas       20480  1 xhci_pci
xhci_hcd              307200  1 xhci_pci
usbcore               323584  8 xhci_hcd,usbnet,usbhid,usb_storage,btusb,xhci_pci,uas,ax88179_178a
```

sebanc, 2021-02-21:
Try kernel 5.10 (replace "/kernel" with "/kernel-5.10" in the grub configuration)

bobafetthotmail, 2021-02-21:
[lsmod-5.10.txt](https://github.com/sebanc/brunch/files/6018273/lsmod-5.10.txt)
[dmesg-5.10.txt](https://github.com/sebanc/brunch/files/6018275/dmesg-5.10.txt)

it still does not detect the USB 3.0 capability of a USB 3.0 hub (which is only detected and used as USB 2.0 hub), nor any other USB 3.0 device, 

Although I was mistaken about the xhci_hcd, it seems to be used in both the 5.4 and 5.10 kernel, if you look at the dmesg entry for the keyboard dongle.
in the 5.4 kernel
```
[  421.326458] usb 1-2.1: new full-speed USB device number 7 using xhci_hcd
[  421.405525] usb 1-2.1: New USB device found, idVendor=046d, idProduct=c52b, bcdDevice=12.10
[  421.405531] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  421.405534] usb 1-2.1: Product: USB Receiver
[  421.405536] usb 1-2.1: Manufacturer: Logitech
```
and in the 5.10 kernel
```
[  508.900080] usb 1-2.1: new full-speed USB device number 10 using xhci_hcd
[  508.982314] usb 1-2.1: New USB device found, idVendor=046d, idProduct=c52b, bcdDevice=12.10
[  508.982331] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  508.982339] usb 1-2.1: Product: USB Receiver
[  508.982346] usb 1-2.1: Manufacturer: Logitech
```

I also noticed that if I use a USB 2.0 extension cable to connect a USB 3.0 device, as a way to "force" it to fallback to USB 2.0 mode now it's detected and works (limited to USB 2.0 bandwith obviously). So I can "work around" the original issue about connecting ethernet dongles at least.

sebanc, 2021-02-22:
Could you try the new brunch-unstable build ?

bobafetthotmail, 2021-02-22:
yes I can try it. I have a spare m.2 drive I can use for testing (I can swap storage drive easily even if it's a tablet). Where do I get this dev build?

sebanc, 2021-02-22:
brunch unstable releases are here:
https://github.com/sebanc/brunch-unstable/releases

bobafetthotmail, 2021-02-24:
the install script of current unstable release (Brunch r88 unstable 20210222) seems broken, fails to write important partition images
```
 Writing partition 1
8.95MiB 0:00:00 [20.9MiB/s] [==========================================================>] 100%            
Writing partition 2
32.0MiB 0:00:00 [ 886MiB/s] [==========================================================>] 100%            
Writing partition 3
2.34GiB 0:00:10 [ 230MiB/s] [==========================================================>] 100%            
Writing partition 4
32.0MiB 0:00:00 [ 495MiB/s] [==========================================================>] 100%            
Writing partition 5
2.34GiB 0:00:10 [ 232MiB/s] [==========================================================>] 100%            
Writing partition 6
 512 B 0:00:00 [3.79MiB/s] [===========================================================>] 100%            
Writing partition 7
pv: ./brunch_r88_20210222/rootc.img: No such file or directory
Writing partition 8
16.0MiB 0:00:00 [ 947MiB/s] [==========================================================>] 100%            
Writing partition 9
 512 B 0:00:00 [4.48MiB/s] [===========================================================>] 100%            
Writing partition 10
 512 B 0:00:00 [3.09MiB/s] [===========================================================>] 100%            
Writing partition 11
8.00MiB 0:00:00 [ 831MiB/s] [==========================================================>] 100%            
Writing partition 12
pv: ./brunch_r88_20210222/efi_secure.img: No such file or directory
ChromeOS installed.
```
and obviously it fails to boot, not finding kernel and such.

Replacing the crhomeos-install.sh with the one from release 87 (the one I have handy) allows to install it.

Apart from that, the unstable release has no usb 3.0 issue.

I also tried a fresh install of 87 and 88, both worked fine. 

So it seems this was just something scewed up in my installation. I did install brunch-toolkit in it and that required to install chromebrew and I also did install Firefox and other stuff in chromebrew. I didn't think adding userspace applicatiosn could blow up USB 3.0 support, maybe that's unrelated, who knows.

Anyway I ran a powerwash on the brunch system that had issues and now it's working fine. Also now brunch-toolkit was updated and does not need to install chromebrew

bobafetthotmail, 2021-02-24:
Thanks for taking the time to follow me through these issues.

bobafetthotmail, 2021-02-24:
wait, spoke too soon, seems that after running an update of brunch and chromeOS with BrunchToolkit

bobafetthotmail, 2021-02-28:
So, after some troubles with latest release (which are unrelated here) I got to test this again.

It seems that the USB 3.0 works correctly only for the FIRST boot after I install ChromeOS from scratch (i.e. I pull the drive, connect it to my OpenSUSE PC and run the script to install stuff on it). From second boot onwards the USB 3.0 does not work anymore and I need to work around it by adding a USB 2.0 cable to force it to use USB 2.0 instead

So the following dmesg and lsmod filew sere collected on a Brunch r88 stable 20210131 with r87 chromeOS, on first boot and after reboot.
[lsmod_log_after_reboot.txt](https://github.com/sebanc/brunch/files/6056155/lsmod_log_after_reboot.txt)
[dmesg_log_after_reboot.txt](https://github.com/sebanc/brunch/files/6056156/dmesg_log_after_reboot.txt)
[lsmod_log.txt](https://github.com/sebanc/brunch/files/6056157/lsmod_log.txt)
[dmesg_log.txt](https://github.com/sebanc/brunch/files/6056158/dmesg_log.txt)