sebanc/brunch #890: how can i get the chromeos r88 recovery image?

wanglrebe, 2021-02-16:
hello
how can i get the chrome os r88 recovery image? the latest version on that two download website is r87

b4iterdev, 2021-02-16:
Download r87 recovery image, then enable native update and update to r88. Make sure you are running brunch r88.

Brax84, 2021-02-16:
after install with r87 recovery

```
ctrl+alt+t
shell
Sudo edit-grub-config
```
add options= enable_updates

close reboot
Then check update

Tash254-ke, 2021-02-17:
> after install with r87 recovery
> 
> ```
> ctrl+alt+t
> shell
> sudo edit-grub-config
> ```
> 
> add options= enable_updates
> 
> close reboot
> Then check update

After enabling native updates, can I disable them and how? because chrome auto updates and this might be an issue with brunch right?

Brax84, 2021-02-17:
After update `sudo edit-grub-config`
delete `options= enable_updates
`
Before update to r89 update Brunch with r99 [releases](https://github.com/sebanc/brunch/releases)
[Update only Brunch](https://github.com/sebanc/brunch#update-only-the-brunch-framework)
[Update Brunch and recovery ChromeOS](https://github.com/sebanc/brunch#update-both-chromeos-and-the-brunch-framework)

Tash254-ke, 2021-02-17:
> After update `sudo edit-grub-config`
> delete `options= enable_updates `
> Before update to r89 update Brunch with r89 [releases](https://github.com/sebanc/brunch/releases)
> [Update only Brunch](https://github.com/sebanc/brunch#update-only-the-brunch-framework)
> [Update Brunch and recovery ChromeOS](https://github.com/sebanc/brunch#update-both-chromeos-and-the-brunch-framework)

Does this work for all options? Like the way you enable them is the same way you disable them?

Brax84, 2021-02-17:
Yes correct

newbit1, 2021-11-02:
Hi,
If the download for the updates is starting,
where exactly is the download saved? I've searched the whole system, but I can't find those file or files from
the download processes.