sebanc/brunch #891: Pixel 2013 (Link) - touchscreen sometimes works, sometimes doesn't

mormislaw, 2021-02-16:
Hi!
Brunch works well on the Chromebook Pixel, using Brunch r88 stable 20210131, after using the correct touchpad config,
but there is one issue - which is the touchscreen.

The thing is, one day, when I turn on the laptop, it works fine, then after waking it from sleep, it stops working. The other day it wouldn't work after turning on, and waking it from sleep fixes the issue. Other times it would work fine all the time, or not work at all.

Is there a way to fix it?

I'd appreciate any help

sebanc, 2021-02-20:
Could you confirm if you still have the issue after installing the latest brunch-unstable release:
https://github.com/sebanc/brunch-unstable/releases

dreit-p, 2021-02-24:
I noticed that after every reboot, you need to touch the screen before you press at least one key or touch the touchpad. After that, the touchscreen works. If you miss this moment, the touchscreen is not available until the restart.

mormislaw, 2021-02-24:
> Could you confirm if you still have the issue after installing the latest brunch-unstable release:
> https://github.com/sebanc/brunch-unstable/releases

Thanks for your answer and help, I'll test it!

mormislaw, 2021-02-24:
> I noticed that after every reboot, you need to touch the screen before you press at least one key or touch the touchpad. After that, the touchscreen works. If you miss this moment, the touchscreen is not available until the restart.

That's curious! I'll see if that's also the case on my laptop. Thanks!