sebanc/brunch #892: is it possible to put kernel 3.12 backports?

djgerman10, 2021-02-18:
is there any way to put kernel 3.12 backports? which supports gma500 driver module

Because there is no system other than windows 7 that supports intel gma3600 graphics

madhits, 2021-02-19:
I have a gma3500 core2duo system using cloudready. So cloudready has this.

sebanc, 2021-02-20:
This is not an issue of gfx drivers but of cpu instructions sets.

Cloudready does not use specific cpu instructions when they build ChromiumOS from source but google does in ChromeOS recovery images.

djgerman10, 2021-02-20:
in cloudready it doesn't work either.

https://guide.neverware.com/technical-details/

sebanc, 2021-02-20:
My point is that it is 100% sure brunch cannot work on core 2 duo.

djgerman10, 2021-02-20:
it's an Atom N2600, I think it's even worse haha

sebanc, 2021-02-20:
Actually Atom 2600 should work but there is a bug with 5.4 kernel, try with kernel 4.19 (refer to the main readme)

closed 2021-02-21 by djgerman10