sebanc/brunch #894: Crosh commands fail and virtual console (chronos) login fails on zork image with intel cpu

justinkb, 2021-02-18:
Using the zork image on my coffee lake + vega 64 desktop pc. Most things are working fine, even linux containers etc. Play store fails to work, but since that might be unrelated to this issue we can put that to the side.

The issue: when switching the the virtual console with ctrl alt f1 and logging in with the default credentials (chronos, no password), all that happens is the terminal briefly goes blank, then it prompts for login again. As if it fails to spawn the login shell or something.

The other issue named in the issue title may be related. When using the in-browser shell crosh, essentially all commands fail to execute. They all output "command failed" or something along those lines. The only command that doesn't produce that error is the help command. That, however doesn't print any help, but it produces no output at all and the command prompt is presented again. Interestingly, tab completion is present in crosh

Let me know if can provide any more information

sebanc, 2021-02-20:
Could you try to see if there are logs in chrome://system ? I would need a dmesg and potentially /var/log/messages file to try to see what the issue is. (even if I think I will not be able to fix it)

khanhtranngoccva, 2021-02-23:
yeet. Use the intel system images (like rammus/hatch/atlas/fizz) instead? Also, make sure NOT to install the ChromeOS image on the Windows system partition, but to repartition the drive and put the image in the other non-system partitions. Otherwise, it would just break.

justinkb, 2021-02-23:
I have to use an image with vega graphics (and zork is the only one), I
don't have any intel integrated gpu

khanhtranngoccva, 2021-02-23:
oof

khanhtranngoccva, 2021-02-23:
You have no integrated graphic cards and an AMD? Sorry for you. Have you tried to install that on a non-system SSD partition? It will not work properly on an HDD because of its slow speed, and not on a Windows partition since the fast startup would interfere.

khanhtranngoccva, 2021-02-23:
Your PC might be quite a good gaming one. Honestly, with such a GPU, I would install BlueStacks/NoxPlayer or the stock Android emulator and have fun with them, as they are optimized for gaming, while ChromeOS has no optimizations or discrete graphics support (none yet officially).