sebanc/brunch #895: Wifi not working after installation

rlug2, 2021-02-19:
Hi! I just installed Brunch on my ThinkPad 10 tablet and all seemed to go well. One I reached the initial set up screen, my wifi does not show any wireless networks. Wifi was working during the installation process when I booted to Linux Mint.

So I searched for solutions and here are the things I tried:
1. Pressed CTRL+ALT+F2 in the ChromeOS screen, logged in as root, then ran sudo edit-grub-config. I added "options=broadcom_wl" after "cros_debug". Pressed CTRL+O then enter, then CTRL+X to exit. Ran sudo reboot. After rebooting, wifi is still not working.
2. Went back to the console, then ran lspci. I saw that it recognized my wireless adapter "01:00.0 Network controller: Broadcom Corporation BCM4356 802.11ac Wireless Network Adapter (rev 02)"
3. Ran rfkill list all, but it only shows "Bluetooth: Soft blocked No, Hard blocked: No". No other network adapters appeared. Tried rfkill unblock all, then rebooted. Wifi is still not working.

I hope you can help me with this issue. Please be patient with me as I am not a programmer, but I can follow instructions. :)

rlug2, 2021-02-19:
By the way, I used brunch_r88_stable_20210131.tar.gz and chromeos_13505.111.0_rammus_recovery_stable-channel_mp-v2.bin.zip. I also uploaded the results of the commands I ran for your reference.
[image]
[image]

sebanc, 2021-02-20:
I have a lot of issues with the broadcom_wl module, I tried to change its source in the recent "brunch-unstable" repo builds:
https://github.com/sebanc/brunch-unstable/releases

Could you try updating to this brunch build and see if it helps ?

rlug2, 2021-02-20:
Thank you! Will try it now and let you know.

rlug2, 2021-02-20:
Hi @sebanc   Wifi is still not working after loading the new build. When I ran rfkill list all, only bluetooth is still showing.

sebanc, 2021-02-20:
I struggle finding accurate info about which brcm driver should be used for which card but from what I read you should not need it, could you remove "options=broadcom_wl" from the kernel command line and post a dmesg ? (run "dmesg > ~/Downloads/dmesg.txt" and attach dmesg.txt from your downloads folder)

rlug2, 2021-02-20:
[dmesg.txt](https://github.com/sebanc/brunch/files/6015112/dmesg.txt)
Hi sebanc. Here's the dmesg.txt file. Thank you!

matrix2113, 2021-02-20:
I'm also running into this issue so if there is anything to solve this issue, I'm using the build on ASUS X205TA

sebanc, 2021-02-21:
Could you try the new "brunch-unstable" release ?
https://github.com/sebanc/brunch-unstable/releases

rlug2, 2021-02-21:
Thank you @sebanc ! Wifi is now working! I am able to get past the set up screen. Thank you very much!

rlug2, 2021-02-21:
Hi @sebanc ! I was able to login and set up my account. I observed some functions that were not working:
- brightness slider is working but brightness does not change. Included options=advanced_als, but nothing changed.
- no option for auto-rotation
- bluetooth turns on but does not detect devices for pairing.

So far these are rhe issues that I observed. Do let me know if I need to open a new issue for these. Thank you very much!

sebanc, 2021-02-21:
For some reason, Brightness slider is not working with the default 5.4 kernel. Try kernel 4.19 or 5.10 (replace "/kernel" with "/kernel-4.19" or "/kernel-5.10" in the grub configuration)

Are you in tablet mode ? auto-rotation will only work if you are in tablet mode.

Fyi, you cannot connect a phone with bluetooth in chromeos, try with audio or input devices.

rlug2, 2021-02-21:
Brightness is now workimy when I changed the kernel to 5.10! Thanks!

As for auto- rotation, it's still not  working even on tablet mode. I just manually rotated  the screen throuigh Display > Orientation > 90°.

For bluetooth, I was trying trying to pair my Bluetooth headsets but they are not detected even if it's is pairing mode. But it's ok since I can use my wired headset.

The camera isn't working as well but I don't mind. 

By the way, my Lenovo Active Pen is working perfectly on the device! 

Thanks a lot for your support!

closed 2021-02-24 by rlug2

Peleryna94, 2025-07-20:
Hi, I had the same problem, using the same firmware and the same brunch. I have the exact same tablet as the author. Wi-Fi works, but Bluetooth doesn't (it sees, detects, and unlocks Bluetooth, but the Wi-Fi itself doesn't see any devices). Is there a way or a brunch that fixed this problem (I currently have an R107)?