sebanc/brunch #896: Touchscreen Issues - Surface 3

rexsububi, 2021-02-20:
Curious if there is any way to incorporate the /linux-surface kernel 5.5+ as an option to support the Surface 3 Laptop? I'm not able to use the downgrade to 4.19. Looking to use IPTSD if possible.

sebanc, 2021-02-20:
ipts kernel module and Iptsd are both included in brunch. Have you added "options=ipts" on the kernel command line ? (in the grub configuration, after cros_debug).

If it is the case, could you post a dmesg ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and post dmesg.txt from your downloads folder).

rexsububi, 2021-02-20:
Oh really? Hm. I should be clear I am running 5.10 for the camera support. Yes, IPTS is enabled. 
[dmesg.txt](https://github.com/sebanc/brunch/files/6015195/dmesg.txt)

Also ran: sudo dmesg | grep ipts
[ 0.000000] Command line: BOOT_IMAGE=(hd0,7)/kernel-5.10 boot=local noresume noswap options=ipts,advanced_als loglevel=7 disablevmx=off cros_secure cros_debug
[ 0.060650] Kernel command line: BOOT_IMAGE=(hd0,7)/kernel-5.10 boot=local noresume noswap options=ipts,advanced_als loglevel=7 disablevmx=off cros_secure cros_debug
[ 1.378488] options=ipts,advanced_als
[ 85.150690] brunch: /rootc/patches/99-bootscripts.sh success
[ 85.337723] brunch: /rootc/patches/99-ipts.sh ipts enabled
[ 85.349726] brunch: /rootc/patches/99-ipts.sh success
[ 91.771623] ipts: loading out-of-tree module taints kernel.
[ 91.788611] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Starting IPTS
[ 91.792661] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Device 045E:0984 ready
[ 92.807472] init: ipts main process (2382) terminated with status 234

rexsububi, 2021-02-20:
Sorry re-ran it with your specific instructions on the placement of options and then rebuilt, ran commands again for the dmesg.

[dmesg9.txt](https://github.com/sebanc/brunch/files/6015671/dmesg9.txt)

[    0.000000] Command line: BOOT_IMAGE=(hd0,7)/kernel-5.10 boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug options=ipts
[    0.060321] Kernel command line: BOOT_IMAGE=(hd0,7)/kernel-5.10 boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug options=ipts
[    1.384034]     options=ipts
[   82.110314] brunch: /rootc/patches/99-bootscripts.sh success
[   82.151885] brunch: /rootc/patches/99-ipts.sh ipts enabled
[   82.158212] brunch: /rootc/patches/99-ipts.sh success
[   88.085945] ipts: loading out-of-tree module taints kernel.
[   88.104057] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Starting IPTS
[   88.108396] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Device 045E:0984 ready
[   89.121686] init: ipts main process (2367) terminated with status 234

sebanc, 2021-02-21:
Could you try the latest "brunch-unstable" release ?
https://github.com/sebanc/brunch-unstable/releases

rexsububi, 2021-02-21:
Post update, IPTS no longer fails and I am able to click on my image for log-in but the screen is non-responsive immediately after. It looks like my model only supports "singletouch option" is that something I should add to grep? I left it on the default kernel and webcam is now resolved without forcing 5.10.

I'm going to try a powerwash, and forcing 5.10 to see if there are any changes. 

[dmesg22.txt](https://github.com/sebanc/brunch/files/6018801/dmesg22.txt)

sudo dmesg | grep ipts
[    0.000000] Command line: BOOT_IMAGE=(hd0,7)/kernel-5.10 boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug options=ipts,enable_updates
[    0.058931] Kernel command line: BOOT_IMAGE=(hd0,7)/kernel-5.10 boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug options=ipts,enable_updates
[    1.420824]     options=ipts,enable_updates
[   85.340984] brunch: /rootc/patches/99-bootscripts.sh success
[   85.420963] brunch: /rootc/patches/99-ipts.sh ipts enabled
[   85.432254] brunch: /rootc/patches/99-ipts.sh success
[   91.186825] ipts: loading out-of-tree module taints kernel.
[   91.201784] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Starting IPTS
[   91.205938] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Device 045E:0984 ready

rexsububi, 2021-02-21:
Powerwash resolved the issue! Thanks for all the help, awesome project.

closed 2021-02-21 by rexsububi