sebanc/brunch #897: Brunch v88 seems not be working on Asus CN60

RagnarLodbrok79, 2021-02-20:
I have tried v88 four times, both vid Chrome 88 and 87.
Evertime I try: I see the white screen with Chrome logo, after that I see a black screen with a mouse cursor. I can move the cursor. When I click the left bottom corner the computer shut off, I can click all the other corners or everywhere without anything happes.

Brunch 87 + CrOS 87 and 88 is good.

sebanc, 2021-02-21:
Could you try the latest brunch-unstable release ?

RagnarLodbrok79, 2021-02-21:
Yes, it works with latest unstable Brunch 88 and CrOS 88.

I got some errors when upgrading Brunch, don't know if it's normal?

[image]

sebanc, 2021-02-21:
This error happens as you are running the script from the downloads folder but it is harmless.

I had to make a change in the new brunch-unstable release and I would need you to verify that it still works on your device. Could you try it ?
https://github.com/sebanc/brunch-unstable/releases

Buktahula, 2021-02-21:
I tried it, everything seems to be working fine (5 minutes test), only Android apps are very slow it feel's like mouse input or screen refresh is not working properly

RagnarLodbrok79, 2021-02-21:
I have also tried it now. It works.
I also have a feeling Android apps are a bit slow.
[image]

RagnarLodbrok79, 2021-02-23:
I'm new to github, should I close this issue now?

RagnarLodbrok79, 2021-03-01:
Now the android apps (lag on mouse input or screen refresh rate) is good as well!

closed 2021-03-01 by RagnarLodbrok79