sebanc/brunch #898: Chrome OS Dev Channel on Brunch r88?

techuser3, 2021-02-21:
I am currently on Chrome OS Stable 88, and I have Brunch r88 too. What would happen if I decided to update from the Stable channel to the Dev Channel in Chrome OS? Would it break Brunch or something like that. I don't want to do it unless it is safe to.

ameyjoshi3068, 2021-02-21:
You can update to the dev channel. It should not break your chrome OS installation.
But, Dev is an experimental channel. You may have stability issues with it, random restarts, lags will most frequently occur.
Dev channel gets an update first, that's why a faulty update may actually break chrome OS.
P.S. I also have a chrome OS dev dual booting setup. I don't use it often and it doesn't have any serious data.

techuser3, 2021-02-21:
Ok thanks I will update to the dev channel right now.

closed 2021-02-21 by techuser3