sebanc/brunch #900: ASUS USB-AC68 Wi-Fi adapter possible with 64bit linux drivers using Brunch?

jordanvanbergen, 2021-02-21:
See: https://www.asus.com/nl/Networking-IoT-Servers/Adapters/All-series/USB-AC68/HelpDesk_Download/ 

I have a ASUS USB-AC68 WiFi-AC1900 adapter with high wireless internet speed. Would love to use this using Chromebooks Brunch releases. 

Do you think it's possible to get the 64bit linux drivers running using Brunch?

sebanc, 2021-02-21:
"options=rtl8812au" should allow it to work

jordanvanbergen, 2021-02-21:
That would be awesome!

I need to upgrade first from an old(er) version (85 I think)  to the latest (88). Forget how to do this so need to look into this first. Will let you know if it will work on 88 release with this option you shared. 

I think if this works I have a new favourite workstation ;-)

jordanvanbergen, 2021-02-22:
So... it's been a while. Have to look into all of this again. 
I am on version: Version 81.0.4044.127 (Official Build) (64-bit)

Am I able to upgrade to 88 in one go?

jordanvanbergen, 2021-02-22:
I have upgraded to Version 87.0.4280.152 (Official Build) (64-bit) 
Brunch: Brunch r88 20210131 

So it's time to try the ASUS USB-AC68 Wireless-AC1900 dongle as you have explained.

jordanvanbergen, 2021-02-22:
I am trying but edit-grub-config does not really work for me because as soon as I use the left,right,up.down buttons on the keyboard all strange characters will be added. 

Is there an alternative way on adding the options value to the grub config? 

So close but so far due to the editor and up,down,right,left keys and strange things happening in the shell within chrome browser. How to enter the terminal outside off the graphical screen? Something with F2 I remember but was not able to get their yet.

jordanvanbergen, 2021-02-22:
I logged in as ROOT and this way the editor is different causing me not to update the edit-grub-config correctly. Doing this with shell and sudo edit-grub-config gave me the editor I as familiar with and I have added the options=rtl8812au behind cros_debug so it's like this now: 

linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug options=rtl8812au

but sadly the USB dongle is not picked/activated and no blue light start on the dongle. So it look like it doesn't work and fall back to the internal wifi that's only able to do 15 Mbps down and 10 Mbps up while this dongle is able to do 400 Mbps / 400 Mbps on my windows machine. 

I really hope I can get this USB dongle working as I really have a splended ChromeOS work station if this is going to work ;-)

Any advise/suggestions I could try?

jordanvanbergen, 2021-02-22:
On this page: http://en.techinfodepot.shoutwiki.com/wiki/ASUS_USB-AC68 it tells the chipset is Realtek RTL8814AU 
Doe Brunch have options=rtl8814au as well to try out?

sebanc, 2021-02-22:
Indeed good catch, try updating to the brunch-unstable release:
https://github.com/sebanc/brunch-unstable/releases

and use "options=rtl8814au"

jordanvanbergen, 2021-02-23:
WoW. Will try this today. 
How do I update to the unstable? Will look in the readme as well. But any suggestion is welcome. 

I wanted to check if the system sees at least 1 USB 3.0 connection. How am I able to check if Brunch/ChromeOS initiates the USB 3.0 as well?

jordanvanbergen, 2021-02-23:
1. chronos@localhost /home/user/3e5ee3b9909628818b083541df41b3e074c3fc77/Downloads $ 
2. sudo chromeos-update -f < path to the Brunch release archive >

Now on version Brunch r88 20210222

Let's see what will happen with the new option ;-)

jordanvanbergen, 2021-02-23:
sudo edit-grub-config 
Added  "options=rtl8814au"

Fingers crossed

jordanvanbergen, 2021-02-23:
THANK YOU VERY MUCH! 

YOU MADE MY DAY!

Still not the full speed I should be able to get but look at this: https://www.speedtest.net/result/10978185956  

100+ Mbps up and down. 
That's enough for browsing and downloading ;-)

jordanvanbergen, 2021-02-23:
Here you see the setup: https://i.imgur.com/0rK3AnE.png 

After working for some minutes it falls back to 10 Mbps again. Will test some other things but the driver is working and that's nice. 

If I know it's really working on USB 3.0 then this issue can be closed and hopefully each new release will have this: "options=rtl8814au" from now on ;-) 

TNX AGAIN!

jordanvanbergen, 2021-02-23:
WoW! Not 100% stable but look: Check out my Speedtest result! How fast is your internet? https://www.speedtest.net/my-result/a/7068937703 

So after reboot I got 165 Mbps in the first minutes ;-)

jordanvanbergen, 2021-02-23:
Even 200+ Mbps on a new test: https://www.speedtest.net/result/10980843574 

So this has become a very nice Chrome OS machine to use for day to day work combining a fast wifi dongle + USB webcam + mic (internal mic still not working) for your every day work routines. 

Only some apps won't work or do what they are supposed to do (like saving Instagram stories). But just to let everyone know that has an old Lenovo IDEAPad 13: https://www.coolblue.nl/product/315005/lenovo-yoga-13-mam3dmh.html it still can be used after 8 years as fast Chrome OS workstation with TOUCH videoscreen, 360 degrees turnable and with the correct USB wifi dongle and USB webcam you have everything you need.