sebanc/brunch #901: how to disable rootfs verification?

Ghaicorp, 2021-02-22:
See https://cloudreadykb.neverware.com/s/article/Disabling-RootFS-Verification

I installed Chrome OS 87 on my Lenovo Ideapad 320 (AMD A9 CPU) and I want to disable rootfs verification so I can view root files via browser and edit the root drive.

Sorry for my English.

sebanc, 2021-02-22:
rootfs verification is disabled by default, you can modify any file you want from crosh shell.

Ghaitech, 2021-02-22:
> rootfs verification is disabled by default, you can modify any file you want from crosh shell.

thanks but i cant access file:/// from the browser

Ghaitech, 2021-02-22:
[image]

see this

sebanc, 2021-02-22:
This feature is not related to rootfs verification and has most likely been blocked in chromeos. I don't really see the point of being able to browse the rootfs files through the browser TBH.

khanhtranngoccva, 2021-02-23:
The root is also read-write in developer mode so don't worry about that.